FreeCAD/FreeCAD #17811: BIM: Add the "Volume" property to the main parameters of architectural components. (Feature request)

kaiwas, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The volume property is very important in architectural design, because this value is often required for calculations and comparison of design options during design.
But now this property has to be calculated using expressions. This is tedious.
Walls, beams, slabs - almost everywhere you have to calculate the volume.

I suggest adding this parameter available by default (like length, height, area, etc.)

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39175 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC.backup1731047833.082574 0.7.4 (Disabled)
  * addFC 0.7.5
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-12:
> Could be fixed by #17438

Thank you!
The screenshot in this topic shows the "Part" details. But if these parameters are everywhere, that would be just great!

yorikvanhavre, 2024-11-13:
BIM objects are also Part objects, so yes, #17438 will solve this issue

paullee0, 2024-11-22:
So, expect a Volume property be added to BIM objects by default when #17438 is merged?

maxwxyz, 2025-01-28:
PR was reverted.

yorikvanhavre, 2025-02-04:
The right way here is probably (as we do with other properties) to check if no "Volume" property exists before adding one, so if/when the physical properties come back, the adding we do at BIM level will simply not happen. But there is also the problem that we need to add code in BIM that updates the Volume property, which should not execute if that is handled by physical properties.

paullee0, 2025-02-04:
Any difference this and other similar properties (e.g. wall area, inner area, footprint, subtraction volume) to be implemented as :-

1. wall.property (document object), or
2. self.property (feature python instance) 

Thanks for comments.

yorikvanhavre, 2025-02-04:
As a document object property. Because internally it's not needed, one can easily retrieve obj.Shape.Volume. This here would essentially be a convenience, read-only property for users to see the object volume

kaiwas, 2025-02-04:
> This here would essentially be a convenience, read-only property for users to see the object volume

Probably, some other setting for dimensions would not hurt.
Volumes most often have to be calculated in cubic meters.
At the same time, the main dimensions in the document are in millimeters.

yorikvanhavre, 2025-02-04:
The volume will take the volume unit that is set in the unit scheme.

The whole thing here was the proposal of the Physical Properties introduced in #17438 . It got reverted because of other issues, but it's still probably going to be added, in one form or another.

So it's maybe wise to wait until executing this one...

kaiwas, 2025-02-04:
> The volume will take the volume unit that is set in the unit scheme.

I don't see any separate settings for linear dimensions/areas/volumes.
There is only a general measurement system.
That is, if I select a system in millimeters, all dimensions will be calculated only in millimeters.
Maybe it's worth creating a separate issue for this? Many CAD systems have separate dimension settings.

furgo16, 2025-02-04:
That caught my eye as well. I only noticed recently that volume seems to be returned always in mm³, regardless of the chosen unit system. Python snippet to illustrate:

```python
doc = App.newDocument()
doc.addObject("Part::Box","Box")
doc.getObject('Box').Length = '10 m'
doc.getObject('Box').Width = '10 m'
doc.getObject('Box').Height = '10 m'
doc.recompute()
obj = doc.getObject("Box")
obj.Shape.Volume

# Always returns 1000000000000.0002, that is, mm³
# Changing project unit system does not change the units of the result
```

If this is the same behaviour we're talking about, perhaps it might be worth filing a separate issue?

yorikvanhavre, 2025-02-05:
Units systems do define their volume unit, though: https://github.com/FreeCAD/FreeCAD/blob/main/src/Base/UnitsSchemaCentimeters.cpp#L42

Maybe you are using one that does not implement a volume unit?

In any case, there should be a broader discussion about unit systems, and how we could migrate out of the hard-coded C++ schemas to give users a bit more flexibility...

furgo16, 2025-02-05:
I'm not an expert on units, so I cannot add much to the discussion. For the sake of completeness, the quick test on my previous comment was done with common unit systems. None of them converted the 1000000000000.0002 mm³ `obj.Shape.Volume` to the expected 1000 m³

- Standard (mm, kg, s, °)
- MKS (m, kg, s, °)
- Building Euro (cm, m², m³)
- Meter decimal (m, m², m³)

I don't discard having done something wrong though, please feel free to correct me if that's the case. I'm happy to file a separate issue for this one too.

yorikvanhavre, 2025-02-05:
If I add a volume property to any object with `obj.addProperty("App::PropertyVolume","Volume")`, that property is correctly set to m³ (or whatever the unit system says). If I attribute a volume to that property, ex: `obj.Volume=obj.Shape.Volume`, the shape volume (174458881.10361594 mm³) is correctly expressed in m³ (0.17 m³)

What error are you getting @furgo16 ?

furgo16, 2025-02-05:
I first noticed it when testing the output of https://github.com/FreeCAD/FreeCAD/pull/19219. Regardless of the unit specified in the BIM schedule, the volume value was always in mm³ (I'm just describing what I observed, that PR did not touch unit systems or units). These two might not be connected, but essentially I'm getting the behavior described in https://github.com/FreeCAD/FreeCAD/issues/17811#issuecomment-2634860270. 

I would have expected that upon changing the unit system, reading `obj.Shape.Volume` would change to m³. But it might well be the wrong expectation by not understanding fully how all this works.

kaiwas, 2025-02-05:
Quick test (on video).
Probably the problem occurs due to the assignment of incorrect property types.
If you output PropertyVolume - then the volume is indicated in the selected dimensions.
But there is no fine-tuning for this. Let me remind you that linear dimensions can be built in millimeters. And areas and volumes can be calculated in meters.
Now, if millimeters are selected - then cubic millimeters are output (not meters).

https://github.com/user-attachments/assets/07e484fa-4ddf-4a07-bb9b-ca7282464291

kaiwas, 2025-02-05:
Example from other programs (some pictures are not in English, but I think the gist is clear)

[image]

[image]

But that's a topic for another issue.

furgo16, 2025-02-05:
Thanks, I learnt something new. So essentially `obj.Shape.Volume` is defined as a float in the code, which means it will always be in mm(³). But if it's assigned to another property defined as a `App::PropertyVolume`, only then the unit conversion is happening as expected.

That means:
- One should never use the raw `obj.Shape.Volume` output directly. It should be assigned to a separate custom property defined as `App::PropertyVolume`
- A potential bug: `obj.Shape.Volume` should be defined in the code as `obj.Shape.Volume` instead of a float. There might be implications here that I don't understand.

Do I understand this correctly?

yorikvanhavre, 2025-02-06:
Yes. Further conclusions:

* App::PropertyVolume is working correctly
* BIM Schedule's volume calculation (unit conversion) is not working correctly

I will have a look at the BIM schedule. Could you check if there are unit conversion problems elsewhere? (Areas, lengths, etc)