FreeCAD/FreeCAD #17812: Spaceball button configuration does not match dialog state after physical button press

PetteriAimonen, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce (requires spacenavd-compatible hardware):

1. Start FreeCAD and create a new document.
2. Right-click toolbar and select `Customize`
3. Open `Spaceball buttons` tab.
4. Click a physical button (other than Button 1) on the controller. This automatically selects the button in the list and correctly shows the currently assigned function. However `Button 1` also stays weirdly half-hilighted:

[image]

5. Switch the assigned function by clicking a new selection in the right side list
6. Close and reopen the dialog
7. Observe that the new function has been assigned to Button 1, while Button 3 function has not changed.

If you select the button in the left list using mouse instead of physical button, the configuration works correctly.

### Full version info

```shell
FreeCAD_weekly-builds-39175-conda-Linux-x86_64-py311.AppImage (also occurs with 1.0RC2)

OS: Ubuntu 22.04.5 LTS (ICEWM/icewm-session/xcb)
Architecture: x86_64
Version: 1.1.0dev.39175 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * dxf_library
  * ThreadProfile 1.84.0
  * Assembly4 0.50.2
```

### Subproject(s) affected?

None

### Anything else?

Related part of code: [DlgCustomizeSpaceball::event](https://github.com/FreeCAD/FreeCAD/blob/45d3864942c6928cec7dc18f95b1f222485756e5/src/Gui/DlgCustomizeSpaceball.cpp#L807). I'm not sure why this occurs though and what this half-selected list state is about.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PetteriAimonen, 2025-08-28:
Still occurs on `FreeCAD_weekly-2025.08.27-Linux-x86_64-py311.AppImage` though something has slightly changed.

```
OS: Ubuntu 22.04.5 LTS (ICEWM/icewm-session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/123.476/1
Installed mods: 
  * dxf_library
  * ThreadProfile 1.84.0
  * Assembly4 0.50.2
```

Default theme has changed so that the half-selected state is no longer visible. The logic is still the same: for modifying the settings it only matters which item I select by mouse, even though the visual selection changes when I press a spacemouse button. Furthermore, in the new build the actual buttons show up as `"Out of range"` while the `"Button 1..32"` have no effect. Assigning a function to an out of range button still works.

[image]

FEA-eng, 2025-10-28:
@PetteriAimonen Can you confirm if this issue is still relevant in the newest weekly build?

PetteriAimonen, 2025-10-28:
Yeah, it still happens the same with `FreeCAD_weekly-2025.10.22-Linux-x86_64-py311.AppImage`:

<pre>
OS: Ubuntu 22.04.5 LTS (ICEWM/icewm-session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/123.476/1
</pre>

pskowronskiTDx, 2025-10-29:
@luzpaz Sorry, I have no capabilities to do any development on Linux at the moment.