FreeCAD/FreeCAD #17814: Freecad breaks when deleting the last feature of a Body used on a FEM Mesh.

leoheck, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad always break when I delete a feature that is linked to another feature.
For instance, using FEM, if i have a mesh linking the last feature of a Body, and then I delete this last feature, freecad breaks. I know this is a problematic operation. But freecad breaking is horrible.

Can we please prevent Freecad of breaking on simple operations like a object deletion?

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
Please provide an example file and steps to reproduce.

FEA-eng, 2024-11-12:
> For instance, using FEM, if i have a mesh linking the last feature of a Body, and then I delete this last feature, freecad breaks.

What do you mean by breaking? Crash or something else? In such a case you should get a warning ("The following referencing objects might break. Are you sure you want to continue?"). Mesh will be outdated and it won't be possible to recreate it because of the missing reference (it may even freeze which is not very nice indeed). Analysis features can lose their references too - FEM itself has no TNP countermeasures. No doubt it's better to mesh Bodies instead of their last features.

leoheck, 2024-11-12:
> Crash or something else

Yes, crash.. The software closes itself immediately.

> In such a case you should get a warning ("The following referencing objects might break. Are you sure you want to continue?").

Nope.

> No doubt it's better to mesh Bodies instead of their last features.

Hm, in this case I believe that being able to apply the mesh on the last feature allows to mesh other internal features.

leoheck, 2024-11-12:
Tried to make a shareable example, and then on this example, it shows me a message with the objects that might break.

Should it include the software itself `FreeCad` in this list? 

Here
[image]

Sample files here (rename it removing the .zip) [fem_mesh_delete_issue.FCStd.zip](https://github.com/user-attachments/files/17722290/fem_mesh_delete_issue.FCStd.zip)

To reproduce this, just delete the `Pocket` that is inside the `Body`, THEN say Yes to see the crash.

commit 98a15bcc (2024-11-22): Fem: Avoid legacy Netgen meshing if there is no referenced shape - fixes #17814
closed 2024-11-22 by chennes