FreeCAD/FreeCAD #17820: Transparency can't have different values accross multiple links

PaddleStroke, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Working on this PR for assembly : https://github.com/FreeCAD/FreeCAD/pull/17818
Where we want to **temporarily** switch some unconcerned objects to transparent to improve visibility.

The problem is for App::Link. 2 links to the same object cannot have different transparency values.
Doing link.Transparency = 80 changes the transparency of the linked object.

I think it is currently possible of doing this with CopyOnChange but for such a temporary change of a large number of objects it is not adapted.

What we need is that App::Link have their own transparency property, the same as they currently have their own 'Visibility' property.

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-11-13:
Link has its own material property. You just need to enable it. It's the view property OverrideMaterial. You can then set the transparency in Link's ShapeMaterial.

PaddleStroke, 2024-11-13:
Oh great thanks. Well this is not an issue then.

closed 2024-11-13 by PaddleStroke

PaddleStroke, 2024-11-13:
@realthunder what about line/point colors ? When I toggle OverrideMaterial, the lines becomes gray and I cannot find a way to change them to match the original color.
So I'm doing the following : 

```
            material = vobj.ShapeAppearance[0]
            material.Transparency = 80
            vobj.ShapeMaterial = material
            vobj.OverrideMaterial = True
```
But then the lines become gray. And there is apparantly no LineColor and PointColors override properties. There is a OverrideColorList that could be it but I can't see how to use it

[image]

PaddleStroke, 2024-11-13:
After some more test it appears that points/lines get the color 'DiffuseColor' of the ShapeMaterial. So it seems they can't be customized so far.