FreeCAD/FreeCAD #17821: Part Desing: Cannot constrain both ends of an Arch Slot

leoheck, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using Arch Slot on Part Design WB, I cannot constraint both ends. 

This is the not fully constrained sketch.
[image]

And when I constraint the second end of an arch slot to a point , I always end up with:
Please, ignore the rotation of the view, here the unconstrained end is on top.
[image]

I never could do this. And I feel it should be possible.
Here the first end is fixed, and I want to constrain the second one so the length is fixed buy the ends. 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
What do you want to constrain? Radius, position of the arc center,... Please provide a demo file.

Roy-043, 2024-11-12:
AFAICT this is possible. See attached [file](https://github.com/user-attachments/files/17722546/arc-slot-both-ends-constrained.FCStd.zip) (remove .zip).

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

leoheck, 2024-11-12:
> What do you want to constrain? Radius, position of the arc center,... Please provide a demo file.

Center points of arcs at the end of a arc slots. On both ends using the coincidence constrain on another point.

leoheck, 2024-11-13:
> AFAICT this is possible. See attached [file](https://github.com/user-attachments/files/17722546/arc-slot-both-ends-constrained.FCStd.zip) (remove .zip).

Indeed... 

Perhaps the problem appears when constraining the center of the Arc with a Point. Tryed to change that in your simple sketch and it worked fine.

But I added my Sketch214 as well... I cannot do the same there.

[arc-slot-both-ends-constrained_2.FCStd.zip](https://github.com/user-attachments/files/17724814/arc-slot-both-ends-constrained_2.FCStd.zip)

Also, the issue may be related with the circle I use as a guide. As soon as I release the Arc Slot, it becomes fully red like this:

[image]

Then If I play a bit with its size, it change colors... 

[image]

There is something strange with this tool.

Roy-043, 2024-11-13:
In your sketch the radii of the slot are defined by Constraint27 (point on edge for the center point of one of the end arcs), Constraint29 (equal constraint for the radius of one of the end arcs) and because the center of the slot is constrained to the origin. Making the center point of the other end arc coincident with one of the point objects would add a redundancy as it again defines the radius of the slot.

To demonstrate the issue:
1. Delete the existing slot.
2. Constrain the center of the large construction circle to the origin.
3. Create an arc slot by picking random points.
4. Constrain the radius of one of the end arcs with an equal constraint as before.
5. Constrain both centers of the end arcs to the point objects.
6. Result: the sketch has only 1 dof. The center of the arc slot can be moved along a diagonal line.
7. Constraining the center of the slot to the origin would remove 2 dofs and therefore create a redundancy.

Potential solutions:
* Apply a vertical constraint between the center of the slot and the origin.
* Remove one of the tangential constraints of the slot, apply a coincident constraint instead, and apply a coincident constrain between the center of the slot and the origin.
* Add a construction circle constrained to the center of the slot and a center of one of the end arcs. Appy an equal constraint between this circle and the existing large construction circle.

[arc-slot-both-ends-constrained_2a.FCStd.zip](https://github.com/user-attachments/files/17728042/arc-slot-both-ends-constrained_2a.FCStd.zip)

maxwxyz, 2024-11-13:
So this is no bug, right?