FreeCAD/FreeCAD #17823: Assembly: Regression: Cannot create a fixed joint in latest main

AIRCAP, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[Fail.zip](https://github.com/user-attachments/files/17721571/Fail.zip)

[image]

**Steps to Reproduce:**

1. Load Fail.FCStd 
OR
1a. Create 2 parts with a cube each
1b. Import both parts into a new, activated Assembly
1c. Agree when to assign a locked constraint to the first part

2. Activate the Assembly
3. Select one face on each of the two cubes
4. Assign a fixed Joint

**Expected Outcome:**
Fixed Joint Dialog appears, Fixed Joint assigned

**Actual Outcome:**
Error Message:
```
Running the Python command 'Assembly_CreateJointFixed' failed:
Traceback (most recent call last):
  File "/mnt/slow1/src/FreeCAD/main/Mod/Assembly/CommandCreateJoint.py", line 105, in Activated
    activateJoint(0)
  File "/mnt/slow1/src/FreeCAD/main/Mod/Assembly/CommandCreateJoint.py", line 62, in activateJoint
    Gui.doCommand(f"panel = JointObject.TaskAssemblyCreateJoint({index})")
  File "<string>", line 1, in <module>
  File "/mnt/slow1/src/FreeCAD/main/Mod/Assembly/JointObject.py", line 1219, in __init__
    self.form.advancedOffsetCheckbox.stateChanged.connect(self.advancedOffsetToggled)

'PySide2.QtWidgets.QWidget' object has no attribute 'advancedOffsetCheckbox'
```

### Full version info

```shell
latest main:

OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39182 (Git)
Build type: Debug
Branch: main
Hash: 5feda62c3a4d1a5d4e4d94262ec6835698d06621
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Defeaturing 1.2.2
  * Silk 0.1.5
  * CurvedShapes 1.0.12
  * CfdOF 1.27.13
  * Curves 0.6.51
  * Plot 2024.10.23
  * AirPlaneDesign 0.4.1
  * lattice2 1.0.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-12:
Cannot confirm on latest weekly
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39175 (Git) Conda
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5
```

I've tried two bodies, to parts, inserting them as link or directly. It works on my side. Also no issue with your file.

CorvusCorax, 2024-11-12:
Apparently this was a build system issue - despite make clean the updated .ui file did not get copied into the build folder - had to completely wipe the build folder and start with a fresh cmake.
closing

AIRCAP, 2024-11-12:
closing because not a bug

closed 2024-11-12 by AIRCAP