FreeCAD/FreeCAD #17824: Part – Create a copy – Refine shape  Works in version 0.19 and fails in 0.20 and 0.21

ClarkP57, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was recently trying to edit a Thingiverse .stl file in order to add mouse-ears to the model. I normally use a different program for this purpose, but I thought I’d try it in FreeCAD. Following the sequence of commands, found in multiple places on Youtube, I was successful using FreeCAD 0.19 and unsuccessful with both versions 0.20.2 and 0.21.2. I thought the following might be helpful.

Thingiverse Thing: Hel-Txt - Standalone Meshtastic communicator (Heltec ESP32 V3_V2)
File generating the error: Heltxt-Cover_for_Silicone_cover

In FreeCAD Version: 0.19 Revision Number 24291
Sequence:
1) Open the above .stl file. – No Errors
2) In the Part Workbench: Create shape from mesh… (default tolerance) - No Errors
3) In the Part Workbench: Convert to solid - No Errors
4) In the Part Workbench: Create a copy > Refine Shape – No Errors

In FreeCAD Version: 0.20.2 Revision Number 29603
4) In the Part Workbench: Create a copy > Refine Shape – Report view: Recompute failed! Please check report view.

In FreeCAD Version: 0.21.2 Revision Number 33771
4) In the Part Workbench: Create a copy > Refine Shape – Report view: Recompute failed! 

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ClarkP57, 2024-11-12:
After some further investigation I've found that in version 0.20.2 if in the "Create shape from mesh..." step, an unchecked check box  titled "Sew shape" appears and the sewing tolerance is greyed out. If the checkbox is left unchecked, everything works correctly and the refine shape step does not generate errors.  The sewing tolerance is something of a mystery to me, in version 0.19 (Branch: releases/FreeCAD-0-19) it is a floating point number, in version 0.19 (Branch: Branch_0.19.4) it is an integer number. In the 0.20 and 21 it's back to floating point again and can be changed if the checkbox is selected. I can't tell if changing this number makes any difference.

mwganson, 2024-11-13:
The file has some non-manifold issues as found when checking using analyze and repair mesh tool in mesh workbench.  I believe the folds on surface erros can be generally ignored.  These non-manifolds are in the area of the letters for PWR and RST.  Checking if it is a solid also shows it not to be one.  Other mesh repair tools could be better suited to the task, then return to FreeCAD with the repaired mesh to work with.  If you want to take the time to learn to use it MeshRemodel addon workbench has some tools to add missing facets.  https://forum.freecad.org/viewtopic.php?t=83786

maxwxyz, 2024-11-13:
Closing this. If there is an issue with an ok file in the latest weekly, please reopen.