FreeCAD/FreeCAD #17825: PartDesign_hole has trouble with a curved surface

CoreyMac, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PartDesign_hole has trouble with a curved surface.  The tool creates 1.5 holes instead of only one.  Also the depth adjustment does not allow the single hole to penetrate both sides of a cylinder.  Similar issues with both 0.21.2 and 1.0RC2.  Example drawing attached.  There are three holes.  One in the base that works fine on the X-Y plane.  The two on each side of the cylinder each create 1 in the middle and an entangled phantom 1/2 of a hole at the edge.  Depth of 200mm does not penetrate the opposite side which is only ~150mm away...
[Outlet Blocker v0.6.FCStd.zip](https://github.com/user-attachments/files/17722172/Outlet.Blocker.v0.6.FCStd.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Here is an illustration of the X-Y relative plane centerlines created and the image of the hole creation results...
Thank you very much for all of your hard work.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-28:
@theo-vt does your latest PR fix this issue ?

theo-vt, 2025-07-28:
I think this person wants the hole to run perpendicular to the axis of the cylinder? In which case no, my PR uses the cylinder's axis as the hole axis, but it does solve the 1.5 hole problem,

thanks!

FEA-eng, 2025-09-27:
Broken preview here:

[image]

The model itself seems to be rather unstable. Maybe because only existing faces are used as profiles for holes.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

CoreyMac, 2025-10-03:
Here is a simple two circle cylinder as described above.

[image]

CoreyMac, 2025-10-03:
Now select the outer surface and try to put a hole through it...  The selection shows the surface selected but the hole is not a hole but a 1/2 of a hole on each end...

[image]

[image]

CoreyMac, 2025-10-03:
Still similar behavior as before though the holes moved a bit.  So a Sketch with two concentric circles fully constrained, one 50mm diameter and one 60mm diameter.

Pad this into a cylinder 100mm tall.

Attempt to put a hole through one or both sides perpendicular to the long axis of the cylinder.  Using the the pocket instead of the hole tool actually thins one wall of the cylinder instead of doping any sort of pocket thing......

Using version:

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DynamicData 2.74.0
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * InventorLoader 1.5.1
  * lattice2 1.0.0
  * parts_library
  * sheetmetal 0.7.22

luzpaz, 2025-12-07:
> Broken preview 

@FEA-eng is this a separate bug ?

FEA-eng, 2025-12-14:
> > Broken preview
> 
> [@FEA-eng](https://github.com/FEA-eng) is this a separate bug ?

@kadet1090 may know that

FEA-eng, 2026-03-15:
The file fails to recompute

```
11:19:13  <Exception> TopoShapeExpansion.cpp(2004): Null shape
11:19:13  Hole001: Null shape
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260303 (Git shallow)
Build date: 2026/03/03 22:57:35
Build type: Release
Branch: grafted,grafted
Hash: 637463ac354f093e14dacfc5059eb70c2b1216fd
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

FEA-eng, 2026-09-06:
[This](https://github.com/FreeCAD/FreeCAD/issues/17825#issuecomment-4062717736) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260720 (Git shallow)
Build date: 2026/07/20 16:51:42
Build type: Release
Branch: grafted,grafted
Hash: 057d51f846f535f4f0b48492e586be0a76b9efe2
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```