FreeCAD/FreeCAD #17827: Crashes with Ubuntu 24.04.1 LTS

G6KHW, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD opens and looks normal. As soon as Itry to open or create a new design the program just closes.
I have tried in safe modewith the same results.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.39175 (Git)
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Anything else?

From syslog. The first lines are the program starting up.
The second set is the program crashing.

2024-11-12T20:38:29.758494+00:00 ian-bultitude-Aspire-E5-551 systemd[2165]: Started app-gnome-freecad\x2ddaily-16515.scope - Application launched by gnome-shell.
2024-11-12T20:38:30.070187+00:00 ian-bultitude-Aspire-E5-551 freecad-daily.desktop[16515]: QSocketNotifier: Can only be used with threads started with QThread
2024-11-12T20:38:30.669567+00:00 ian-bultitude-Aspire-E5-551 freecad-daily.desktop[16515]: QOpenGLFunctions created with non-current context
2024-11-12T20:38:33.671600+00:00 ian-bultitude-Aspire-E5-551 gnome-shell[2427]: surface_state_changed: assertion 'wl_window->has_last_sent_configuration' failed

2024-11-12T20:38:54.703727+00:00 ian-bultitude-Aspire-E5-551 freecad-daily.desktop[16515]: freecad-daily: ./src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
2024-11-12T20:38:56.170620+00:00 ian-bultitude-Aspire-E5-551 systemd[2165]: app-gnome-freecad\x2ddaily-16515.scope: Consumed 6.969s CPU time.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-12:
I am using this same OS inside a virtual box virtual machine with a windows 10 host.  I had to set the environment variable suggested in that error message.  I believe I added this line to the bottom of ~/.bashrc

`export COIN_GL_NO_CURRENT_CONTEXT_CHECK=1`

But I might have needed it elsewhere, too.  I don't remember exactly the details anymore.

G6KHW, 2024-11-13:
Hi Mark,
Thank you for the reply,
I’ll have another look at that.
Regards
Ian

G6KHW, 2024-11-13:
Ok Mark,
I tried that and it made no difference.
Regards Ian

mwganson, 2024-11-13:
If nothing changed, then you're still the same error message, which means the system isn't seeing the environment variable.

I think my instructions were not clear enough.  If you add 

`export COIN_GL_NO_CURRENT_CONTEXT_CHECK=1`

to the bottom of ~/.bashrc, then that only provides the environment variable to applications run from the terminal.  This is how I run my self-compiled version in order to test out PR's before submitting them.  You are probably running FreeCAD from the Gui, so this doesn't work so well.  When adding to /etc/environment, then "export" should be removed:

`COIN_GL_NO_CURRENT_CONTEXT_CHECK=1`

Then you can restart or use:

`source /etc/environment`

to load in the new environment variable.

sudo nano /etc/environment
copy/paste in the above line without the export keyword
exit and save changes
restart or use the source option above.

mwganson, 2024-11-13:
I'm happy to walk you through it.  Describe in detail the steps you have taken to follow my above instructions.  Perhaps I will be able to see where you went astray.

G6KHW, 2024-11-13:
Update:-

So I have removed the edit from ~/.bashrc
Made the edit to /etc/environment as mentioned
Rebooted

and tried the FreeCAD again
This time it does appeared to have resolved the imediate issue.

Thank you.

Should this not be carried out as part of the installation process?

mwganson, 2024-11-13:
Perhaps it should be part of the install process.  I don't know how often people run into it.  I figured i ran into it because of my virtual machine configuration and because I was self-compiling.  I still get loads of warnings, but no crashes.

G6KHW, 2024-11-22:
> @mwganson do you want to propose a PR for this edgecase ?

Sorry I do not understand what a "PR" or "edgecase" is.
As yet I have not tested the newly released version 1.0.

luzpaz, 2024-11-22:
> Sorry I do not understand what a "PR" or "edgecase" is.

@G6KHW that question was directed toward @mwganson  
But, JFYI: 
* [Pull Request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests#:~:text=A%20pull%20request%20is%20a,changes%20into%20the%20main%20codebase.) 
* 'edgecase' [definition](https://en.wikipedia.org/wiki/Edge_case)

>  As yet I have not tested the newly released version 1.0.

That would be nice but not necessary. Just as long as you're running and testing against the latest development build is what is important.

G6KHW, 2024-11-22:
Thank you for the reply however, I am only a new user and not a programmer or developer, therefore, do not know if I am expected to do anything.
I found an issue which I reported and was given a work-around.
I must remove this and try the new version when I have some time (hopefully this week-end)

luzpaz, 2024-11-22:
> Thank you for the reply however, I am only a new user and not a programmer or developer, therefore, do not know if I am expected to do anything.

@G6KHW you've done a lot so far and thank you for all the steps you've taken. You opened a ticket and so discussions about how to mitigate the issue in FreeCAD you presented are being discussed. If there is something that someone needs from you, they will directly ping you using `@` in front of your github user name G6KHW. Like so @G6KHW

G6KHW, 2024-11-22:
Hi All,
Many thanks for the info etc.

I am happy to report that the new version does not require the edited environment setting.as mwganson had me make.
I removed this and tested the Freecad-Daily to confirm the crash then updated the software (which unfortunately removed the version 0.21) and was unable to repeat the crash senecio that I experienced before.

Thank you and well done.

luzpaz, 2024-11-22:
@G6KHW Thanks for the follow-up. Can you add the full [about](https://wiki.freecad.org/About) info you tested this on ?

G6KHW, 2024-11-22:
About as requested:-

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

closed 2024-11-22 by luzpaz