FreeCAD/FreeCAD #17828: freecad-1.0rc2 build correctly but crash when execute with PySide-6.8.0

Tarot67, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I update my system and passed to QT6.8.0 i just build PySide6.8.0 freecad build without problem as usual.
But when freecad is executed it crash with libshiboken6 object error. I don't know why.

### Full version info

```shell
Libs: 1.0.0RC2R39077 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x430d0) [0x7f05812430d0]
#1  /usr/lib64/libpython3.11.so.1.0(PyObject_GetAttr+0x12) [0x7f058480b892]
#2  /usr/lib64/libshiboken6.abi3.so.6.8(+0x1a04b) [0x7f0583b2104b]
#3  /usr/lib64/libshiboken6.abi3.so.6.8(+0x1a288) [0x7f0583b21288]
#4  /usr/lib64/libshiboken6.abi3.so.6.8(InitSignatureBytes+0x1b) [0x7f0583b24a2b]
#5  /usr/lib64/python3.11/site-packages/PySide6/QtCore.abi3.so(+0x2cb29e) [0x7f054c8cb29e]
#6  /usr/lib64/libshiboken6.abi3.so.6.8(+0x3406b) [0x7f0583b3b06b]
#7  0x7f0583b3b34d in Shiboken::Module::resolveLazyClasses(_object*) from /usr/lib64/libshiboken6.abi3.so.6.8+0x7d
#8  /usr/lib64/libshiboken6.abi3.so.6.8(+0x34485) [0x7f0583b3b485]
#9  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#10  /usr/lib64/libpython3.11.so.1.0(PyObject_Call+0x9c) [0x7f05847e6c7c]
#11  /usr/lib64/libshiboken6.abi3.so.6.8(+0x167d6) [0x7f0583b1d7d6]
#12  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#13  /usr/lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f05847e57fb]
#14  /usr/lib64/libpython3.11.so.1.0(+0x17268a) [0x7f058477268a]
#15  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#16  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0x9a) [0x7f05848bd6ea]
#17  /usr/lib64/libpython3.11.so.1.0(+0x2bc7a5) [0x7f05848bc7a5]
#18  /usr/lib64/libpython3.11.so.1.0(+0x20a666) [0x7f058480a666]
#19  /usr/lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x70d4) [0x7f0584849284]
#20  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#21  /usr/lib64/libpython3.11.so.1.0(+0x1e624e) [0x7f05847e624e]
#22  /usr/lib64/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0xff) [0x7f05847e711f]
#23  /usr/lib64/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x3ac) [0x7f058485c64c]
#24  /usr/lib64/libpython3.11.so.1.0(+0x23fe5b) [0x7f058483fe5b]
#25  /usr/lib64/libpython3.11.so.1.0(+0x20a666) [0x7f058480a666]
#26  /usr/lib64/libshiboken6.abi3.so.6.8(+0x34415) [0x7f0583b3b415]
#27  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#28  /usr/lib64/libpython3.11.so.1.0(PyObject_Call+0x9c) [0x7f05847e6c7c]
#29  /usr/lib64/libshiboken6.abi3.so.6.8(+0x167d6) [0x7f0583b1d7d6]
#30  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#31  /usr/lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f05847e57fb]
#32  /usr/lib64/libpython3.11.so.1.0(+0x17268a) [0x7f058477268a]
#33  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#34  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0x9a) [0x7f05848bd6ea]
#35  /usr/lib64/libpython3.11.so.1.0(+0x2bc7a5) [0x7f05848bc7a5]
#36  /usr/lib64/libpython3.11.so.1.0(+0x20a666) [0x7f058480a666]
#37  /usr/lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x70d4) [0x7f0584849284]
#38  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#39  /usr/lib64/libpython3.11.so.1.0(+0x1e624e) [0x7f05847e624e]
#40  /usr/lib64/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0xff) [0x7f05847e711f]
#41  /usr/lib64/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x3ac) [0x7f058485c64c]
#42  /usr/lib64/libpython3.11.so.1.0(+0x23fe5b) [0x7f058483fe5b]
#43  /usr/lib64/libpython3.11.so.1.0(+0x20a666) [0x7f058480a666]
#44  /usr/lib64/libshiboken6.abi3.so.6.8(+0x34415) [0x7f0583b3b415]
#45  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#46  /usr/lib64/libpython3.11.so.1.0(PyObject_Call+0x9c) [0x7f05847e6c7c]
#47  /usr/lib64/libshiboken6.abi3.so.6.8(+0x167d6) [0x7f0583b1d7d6]
#48  /usr/lib64/libpython3.11.so.1.0(+0x20a7bd) [0x7f058480a7bd]
#49  /usr/lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x6b) [0x7f05847e57fb]
#50  /usr/lib64/libpython3.11.so.1.0(+0x17268a) [0x7f058477268a]
#51  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#52  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0x9a) [0x7f05848bd6ea]
#53  /usr/lib64/libpython3.11.so.1.0(+0x2bc7a5) [0x7f05848bc7a5]
#54  /usr/lib64/libpython3.11.so.1.0(+0x20a666) [0x7f058480a666]
#55  /usr/lib64/libpython3.11.so.1.0(PyObject_Vectorcall+0x33) [0x7f05847e5fc3]
#56  /usr/lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x5b9) [0x7f0584842769]
#57  /usr/lib64/libpython3.11.so.1.0(+0x2413ac) [0x7f05848413ac]
#58  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0x9a) [0x7f05848bd6ea]
#59  /usr/lib64/libpython3.11.so.1.0(+0x2d171d) [0x7f05848d171d]
#60  /usr/lib64/libpython3.11.so.1.0(+0x2d16aa) [0x7f05848d16aa]
#61  /usr/lib64/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f05848d1619]
#62  0x7f0585541ae9 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/local/lib64/libFreeCADBase.so+0xed
#63  0x7f058807e49b in Gui::Application::runInitGuiScript() from /usr/local/lib64/libFreeCADGui.so+0x3d
#64  0x7f058824dcfa in Gui::StartupPostProcess::showMainWindow() from /usr/local/lib64/libFreeCADGui.so+0x64
#65  0x7f058824c6c2 in Gui::StartupPostProcess::execute() from /usr/local/lib64/libFreeCADGui.so+0x9c
#66  0x7f058807f279 in Gui::Application::runApplication() from /usr/local/lib64/libFreeCADGui.so+0x15d
#67  FreeCAD() [0x428487]
#68  /lib64/libc.so.6(+0x29d57) [0x7f0581229d57]
#69  /lib64/libc.so.6(__libc_start_main+0x85) [0x7f0581229e15]
#70  FreeCAD() [0x427721]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I fix problem with installation of PySide6.8.0.2 with recompilation of freecad. I think probably a incompatibility binary or bad binary indexation something like that! i don't know really. I reporting this because if anybody has this problem update PySide6 to 6.8.0.2.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-13:
Is this also happening with RC3 or latest main?
@adrianinsaval FYI

adrianinsaval, 2024-11-13:
what is your system? It is not clear to me if you have fixed this or not.

adrianinsaval, 2024-11-13:
If I understood correctly what you are saying, this is an upstream issue that got fixed in 6.8.0.2, possibly with https://invent.kde.org/qt/pyside/pyside-setup/-/commit/2e606c31dfc66b687b1483a9f66dcc61cd880d2e, any distro shipping 6.8.0 is most likely going to ship the 6.8.0.2 update so I don't think there is any action we need to take on our side

Tarot67, 2024-11-13:
Yes no actions but it IS juste for information because i paased a lot of
Time and thé problème came from pyside-6.8.0 and not from Freecad

closed 2024-11-13 by adrianinsaval