FreeCAD/FreeCAD #17829: groving helix (end of thread) resulting in broken face

atapi, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When cutting end of inner threads created with helix (tapering edges) using grove, sometimes resulting in broken inner thread face.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.30
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Anything else?

1. I created simple circle pad with hole inside
[image]

2. Created ACME-like thread with helix
[image]
[image]

3. Next tried to cut one side for tapered end - and result was ok.
[image]

4. Tried to do the same on other end by mirroring - results with error, broken face (selected).
[image]
[image]

I've tried also making independent sketch and grove other side - but result are the same.
Sometimes with specific cutting profile, cut completes without problem, but mostly this problem exist.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

atapi, 2024-11-12:
Also, when thread profile do not starts at bottom of hole (eg. 1mm higher) cutting completes without problem.

maxwxyz, 2024-11-13:
@atapi can ypu please provide an example file and also try with RC3 or the weekly version:  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

@FlachyJoe @AIRCAP FYI

atapi, 2024-11-13:
I tried weekly-builds from last monday - result the same.

Example file from description above:
[error helixcut.zip](https://github.com/user-attachments/files/17735425/error.helixcut.zip)

When you change "Sketch002" Attachment Offset->Position->y to 5mm - mirrored grove cuts without error.

FlachyJoe, 2024-11-13:
It works if you switch off the AdditiveHelix Refine property.

atapi, 2024-11-13:
Yes, it works, but only after disablling Refine also in first groving operation.
When I disabled Refine on AddativeHelix only - second grove operation still produces error.