FreeCAD/FreeCAD #17830: Assembly selection problems related to linked components. Lots of [TopoShapeExpansion.cpp(970): hasher mismatch] errors

MiqCG, 2024-11-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As commented in [#17033](https://github.com/FreeCAD/FreeCAD/issues/17033) here I open a new issue to split the previous one, as the case that I reported in the 11th of October needs to be addressed separately.

Case:
I have made a new part (https://github.com/FreeCAD/FreeCAD/pull/110-006), with 1.1.0-38898, and when I tried to add to an assembly, most of the geometric part weren’t recognized as a usable JCS. On the Report view, lots of " TopoShapeExpansion.cpp(970): hasher mismatch" and "<class 'IndexError'>" when trying to select a geometry.

See screencast attached and files, just in case the part has been badly created for some reason. In the folder there is several test of assembly (inserted or linked, copied, and with RC2 revision).
Test made with clean .config .cache. share folders. And same behaviour with revision 1.0.0RC2.

Link to original comment:
https://github.com/FreeCAD/FreeCAD/issues/17033#issuecomment-2406166845

@furgo16 and @CalligaroV , let me know if the title or the description are incorrect or insufficient.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Assembly

### Anything else?

[Ass_test_1.zip](https://github.com/user-attachments/files/17723966/Ass_test_1.zip)

https://github.com/user-attachments/assets/b9273281-639a-44ca-9e0a-4177c91c3fa0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2024-11-12:
Testing my own files in revision 39175 I have found at least one problem in the #110-006.FCStd file attached: the tip wasn't on the last feature. Not the first time I make this basic mistake; definitely I need urgently a far more obvious way to represent the tip in the Tree. 
BUT even when the tip is situated in the last operation (Mirrored), hundreds of
`<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch`
errors continue appearing. 

Can't dismiss further file construction problems but I couldn't find them.

Reattached files with the tip moved:
[Ass_test_1_corrected.zip](https://github.com/user-attachments/files/17724064/Ass_test_1_corrected.zip)

maxwxyz, 2024-11-14:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/17554

MiqCG, 2024-11-14:
I don't think this issue is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/17554 because on 1.0RC2, the WheelAxle assembly doesn't output any error on the Report View when going over the components with the mouse.
And as I said in my second comment: 
> even when the tip is situated in the last operation (Mirrored), hundreds of
<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch
errors continue appearing.

maxwxyz, 2024-11-15:
Could you retest on RC4 or latest weekly?

MiqCG, 2024-11-15:
With the last operation visible and with the tip on it, same result with 1.0.0RC4 and with 1.1.0-39175 (almost the last weekly): hundreds of `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` errors in the report view when hovering the mouse over the 110-006 component to attach a marker or JCS (fixed) on it. Maybe it is related to a particular operation in the body or the presence of the datum planes. In case it gives any clue, the part was created in the 10th of October.

No errors on opening the files, recomputing (Ctrl-r) or solving the assembly before trying to create the joint.

furgo16, 2024-11-16:
@MiqCG thanks.

> `@furgo16` and `@CalligaroV` , let me know if the title or the description are incorrect or insufficient.

Regarding the description, I would suggest to add the steps to reproduce to the description. You can copy-paste the ones I put there: https://github.com/FreeCAD/FreeCAD/issues/17033#issuecomment-2406536987

Just think about it from the POV of a developer, a triager or a tester that has to process dozens of issues in a day: being welcomed with a wall of text and a video to watch will take them extra time to process. You have all the context of your issue, but someone new to it won't. Having clear steps on how to reproduce the issue and the outcome will help cutting down the time.

MiqCG, 2024-11-16:
@furgo16 , thanks for your indications. 
With the first comment of this issue I was trying, expressly, not to repeat information as I believe that this issue can only be understood in the context and evolution of the parent issue [#17033.](https://github.com/FreeCAD/FreeCAD/issues/17033)
On the other hand, I thought it was an obvious follow up of @CalligaroV work, and then wasn't needed to re-contextualize. 
Anyway, I will try to be more explicit and verbose the next time.

CalligaroV, 2024-11-16:
@MiqCG, as anticipated in my https://github.com/FreeCAD/FreeCAD/issues/17033#issuecomment-2480596743, I tested your [Ass_test_1_corrected.zip](https://github.com/user-attachments/files/17724064/Ass_test_1_corrected.zip) with my branch https://github.com/CalligaroV/FreeCAD/tree/toponaming-ElementMapVersion-code-from-LS3 (rebased locally on top of 5feda62c3a4d1a5d4e4d94262ec6835698d06621).

The `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` warnings still triggers but IMO the overall output is correct:

[image]

However, I'm a little bit confused: are your tests still giving the same result of the video in the OP also with newer builds (RC4 or newer)?
Only to understand if the modified code in my branch is doing anything good or if the output in my video is due to other merged PRs.

Regardless for that, to (hopefully) completely solve both this issue and #17033 (the `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` warnings) more modifications / imports from LS3 are needed.

@furgo16, in general I would say that I completely agree with your https://github.com/FreeCAD/FreeCAD/issues/17830#issuecomment-2480458543.
For this case however I would say that we're lucky as the issue isn't that hard to reproduce: for me it was enough to follow the steps of the video in the OP.
The best, IMO, is to copy-paste the list of commands recorded by the Python console as it exposes what FreeCAD does under the hood.

Besides that, @maxwxyz I agree with @MiqCG: IMHO this issue, if still reproducible with RC4 or newer builds, should stay opened according to @furgo16's https://github.com/FreeCAD/FreeCAD/issues/17033#issuecomment-2471775336

> I'd favor opening a separate issue for @MiqCG's bug for incremental improvement reasons, and to make it easier to debug and test. Also,
> 
> > if we consider what the 2 cases have in common (the  TopoShapeExpansion.cpp(970): hasher mismatch warning messages) then I'd say that the issue is the same even if AFAICT the root cause (in the code) of those messages is different.
> 
> Along those lines, if the root cause is different, then I would guess that it would require a separate PR for each cause, each with its own test case.

If we don't agree on that it's fine for me but IMO it would be more correct at least to mark this one as a duplicate of #17033

Thanks to everyone!

MiqCG, 2024-11-16:
> However, I'm a little bit confused: are your tests still giving the same result of the video in the OP also with newer builds (RC4 or newer)?

Yes, @CalligaroV the attached files in the original post, tested in RC4 and in dev39175 are outputting the same insane amount of `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch`errors.

If it helps, I have remade a new version of the component '#110-006 in case there was a problem with the revision with which it was made (new file named '#110-006-RC4). Same errors occur when the assembly is replicated with links (insert component tool) and the mouse is passed over the component when creating a fixed joint (Assembly_link_RC4 file). 
No errors are send when the assembly is created copying the components directly inside the assembly file (Assembly_drag_RC4 file).

Then I thought that the problem was in one of the operations in the component '#110-006, and I have tested the assembly deleting the last operation at a time. I have proceeded until I got only the first Pad but the errors keep coming110-006-RC4-nM file) . So I don't know what's the problem because the first operation is a pretty normal extruded sketch.

[Ass_test_1_RC4.zip](https://github.com/user-attachments/files/17787398/Ass_test_1_RC4.zip)

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

CalligaroV, 2024-11-17:
@MiqCG many thanks for the added info.

I tested the assemblies inside [Ass_test_1_RC4.zip](https://github.com/user-attachments/files/17787398/Ass_test_1_RC4.zip) and I can confirm your analysis: the `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` warning messages triggers with `Assembly_link_RC4.FCStd` and `Assembly_link_RC4_nM.FCStd` (that uses `110-006-RC4-nM.FCStd`) but not with `Assembly_drag_RC4.FCStd`

This makes me confident enough to say that this issue is caused by the presence of **external files** in the assembly.
I would suggest to add this in the issue title.

TIA!

MiqCG, 2024-11-17:
> caused by the presence of **external files** in the assembly.
> I would suggest to add this in the issue title.

Done

furgo16, 2024-11-17:
>This makes me confident enough to say that this issue is caused by the presence of external files in the assembly.
I would suggest to add this in the issue title.

This seems to be in common with the other issue, then: https://github.com/FreeCAD/FreeCAD/issues/17033#issuecomment-2407574023

CalligaroV, 2024-11-25:
@MiqCG @furgo16, a quick update:

During the past weekend I tried to import a small part of the code regarding `PartDesign::Transformed` (from which derives `PartDesign::Mirrored`, the last feature of `110-006-RC4.FCStd` and source of the various `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` warning messages) but this doesn't look enough, I still get those warning messages.

At this point I think that a bulk import of the missing calls to `getDocument()->getStringHasher()` from LS3 is the fastest thing to do and will hopefully prevent future issues like this one.

Unfortunately I can't give you an ETA but most likely in the next weekend, or in the second half of this week, I may complete the import.

I'll keep you updated.

furgo16, 2024-12-01:
@CalligaroV thanks so much.

Now that you're talking about the Assembly workbench, this comment from realthunder in the context of Assembly 3 might also be relevant:

> Hashing mismatch error is cause by the external link, which brings a shape from another document with a different hasher than the linking document. This will be fixed in the next release.

Perhaps looking at how Assembly 3 fixed this, instead of looking at LS3 might be another avenue to explore?

Source: https://forum.freecad.org/viewtopic.php?p=228949#p228949

CalligaroV, 2024-12-08:
Thank you too @furgo16 for the added info!

I'll do my best to figure out what can be used from the links you shared.

Sadly ATM I can't (again) give you and ETA for another reply but I'll try to be back in the next weeks (hopefully Christmas Holidays will give me more time to invest in FreeCAD development)

Sorry for the delay!

furgo16, 2024-12-08:
No worries, thank you!

furgo16, 2025-01-13:
@CalligaroV friendly reminder in case you've had some time to look into [the comment above](https://github.com/FreeCAD/FreeCAD/issues/17830#issuecomment-2510246774). Or perhaps @realthunder might have some insight on the "hasher mismatch" errors? Thanks!

01000001khan, 2025-05-28:
I've searched for a few hours and this seems to be the most relevant to my own issue. I downloaded FreeCAD yesterday, so it should be up to date. In this assembly file, when I open the joint tool for any tool, I get the error `<TopoShape> TopoShapeExpansion.cpp(970): hasher mismatch` for every frame that the mouse is moving over the frame part. It took me a while to track down the issue to just these two parts, but I'll see if I can make an even smaller problematic file
[bug.zip](https://github.com/user-attachments/files/20492255/bug.zip)

drwho495, 2025-07-27:
please try in 1.1dev

i commited a fix months ago that fixed this

maxwxyz, 2025-07-27:
@MiqCG  https://github.com/FreeCAD/FreeCAD/releases/weekly-2025.07.26/

luzpaz, 2025-12-11:
@MiqCG ping  
@01000001khan ping