FreeCAD/FreeCAD #17832: appimige 1RC3 not run

RomanOst2020, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

appimige 1RC3 not run.: FreeCAD_1.0.0RC3-conda-Linux-x86_64-py311.AppImage . 
RC2 - ok run. RC1 - not run.

### Full version info

```shell
appimige 1RC3 not run
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

katepavlu, 2024-11-13:
Can you provide more info?
- what distro are you using
- what is the output when you try to run the program in the terminal? 
  - steps to do this:

1.   Open terminal
2. `cd ~/Downloads` (or your language equivalent) 
3. `./FreeCAD_1.0.0RC3-conda-Linux-x86_64-py311.AppImage`

mhioz, 2024-11-13:
It does not run on my PC (running Ubuntu 24.04.1 LTS) either.
The output from the terminal is:
`execv error: No such file or directory`

katepavlu, 2024-11-13:
Are you sure the file is marked as executable?
Right click it, go to Properties, Permissions, check "Allow executing file as program"

katepavlu, 2024-11-13:
> It does not run on my PC (running Ubuntu 24.04.1 LTS) either. The output from the terminal is: `execv error: No such file or directory`

this error just means that you either saved the file somewhere else than in Downloads, or that the name is different

mhioz, 2024-11-13:
File permissions are set correctly. When I move the FreeCAD_1.0.0RC3 AppImage back to the download folder, it does not work either.
The FreeCAD_1.0.0RC2 AppImage in the same folder and with the same file permissions works, but RC3 does not.

mhioz, 2024-11-13:
I got it running. It was related to an AppImage helper software (AppImageLauncher). After removing the launcher it works.

katepavlu, 2024-11-13:
@RomanOst2020 can you confirm that you have permissions set correctly and that you are not using AppImage Launcher?
I tested the appimage on ubuntu 24.04 (in a VM) and it runs fine for me

longrackslabs, 2024-11-14:
Neither RC3 or RC4 AppImage will load for me (RC2 runs fine). I use (and always have) AppImage launcher.

My setup:

System:
  Kernel: 5.15.0-124-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: GDM3 42.0, LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 55% (should be ignored) rechargeable: yes status: Discharging
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3 rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 2605 high: 3465 min/max: 2200/3800 boost: enabled cores: 1: 2765 2: 3465
    3: 2228 4: 2239 5: 2227 6: 2610 7: 2235 8: 2227 9: 2796 10: 3290 11: 2776 12: 2331 13: 2895
    14: 2873 15: 2499 16: 2233 bogomips: 121605
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: nvidia v: 535.183.01 pcie:
    speed: 8 GT/s lanes: 16 ports: active: none off: HDMI-A-1 empty: DP-1,DVI-D-1,DVI-I-1
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 122 s-size: 533x302mm (21.0x11.9") s-diag: 613mm (24.1")
  Monitor-1: HDMI-0 res: 2560x1440 hz: 60 dpi: 124 size: 526x296mm (20.7x11.7")
    diag: 604mm (23.8")
  OpenGL: renderer: NVIDIA GeForce GTX 950/PCIe/SSE2 v: 4.6.0 NVIDIA 535.183.01
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-124-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
  IF-ID-1: docker0 state: down mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:4
    chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e hci-v: 4.0 rev: 1000
Drives:
  Local Storage: total: 1.95 TiB used: 970.97 GiB (48.7%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: 111130WD temp: 53.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s type: N/A serial: <filter> rev: 3B01
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s type: HDD rpm: 7200 serial: <filter> rev: 1A01 scheme: MBR
Partition:
  ID-1: / size: 365.24 GiB used: 109.31 GiB (29.9%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 430.81 GiB (47.0%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 43 MiB (0.0%) priority: -2
    dev: /dev/nvme0n1p3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 10 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
[... 30 lines cut ...]
  Active apt repos in: /etc/apt/sources.list.d/docker.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/docker.gpg] https: //download.docker.com/linux/ubuntu jammy stable
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/kicad-kicad-8_0-releases-jammy.list
    1: deb [signed-by=/etc/apt/keyrings/kicad-kicad-8_0-releases-jammy.gpg] https: //ppa.launchpadcontent.net/kicad/kicad-8.0-releases/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
[... cut ...]
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Processes: 473 Uptime: 6d 14h 49m wakeups: 407 Memory: 62.72 GiB used: 17.59 GiB (28.0%)
  Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 clang: 14.0.0-1ubuntu1.1
  Client: Cinnamon v: 6.0.4 inxi: 3.3.13

katepavlu, 2024-11-14:
@longrackslabs 

AppimageLauncher hasn't had a release in the last four years. This could very well be a bug in that, instead of FreeCAD. I would personally suggest using [gear lever](https://flathub.org/apps/it.mijorus.gearlever) instead, can you try if that one works on your system? (it does on mine).

longrackslabs, 2024-11-14:
Thanks @katepavlu , looks like i am a Gear Lever user now!