FreeCAD/FreeCAD #17833: Assembly: 1.0.0RC3 cannot create Revolute Joint: advancedOffsetCheckbox issue.

Roy-043, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Similar to: #17823.

Error message:
```
Running the Python command 'Assembly_CreateJointRevolute' failed:
Traceback (most recent call last):
  File "D:\BKG_Tmp\FreeCAD_1.0.0RC3-conda-Windows-x86_64-py311\Mod\Assembly\CommandCreateJoint.py", line 129, in Activated
    activateJoint(1)
  File "D:\BKG_Tmp\FreeCAD_1.0.0RC3-conda-Windows-x86_64-py311\Mod\Assembly\CommandCreateJoint.py", line 61, in activateJoint
    panel = TaskAssemblyCreateJoint(index)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\BKG_Tmp\FreeCAD_1.0.0RC3-conda-Windows-x86_64-py311\Mod\Assembly\JointObject.py", line 1244, in __init__
    self.adaptUi()
  File "D:\BKG_Tmp\FreeCAD_1.0.0RC3-conda-Windows-x86_64-py311\Mod\Assembly\JointObject.py", line 1487, in adaptUi
    advancedOffset = self.form.advancedOffsetCheckbox.isChecked()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'PySide2.QtWidgets.QWidget' object has no attribute 'advancedOffsetCheckbox'
```

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc3)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-11-13:
There seem to be serious regressions with joints in Assembly pretty much breaking it:
https://forum.freecad.org/viewtopic.php?t=92063

Other:
https://github.com/FreeCAD/FreeCAD/issues/17694
https://github.com/FreeCAD/FreeCAD/issues/17556

PaddleStroke, 2024-11-13:
This error here `'PySide2.QtWidgets.QWidget' object has no attribute 'advancedOffsetCheckbox'` is as Max mentioned due to .ui files changed. So this just needs a clean build to fix.
Since Roy does not build I seem to recall, it'd mean there was a build failure in some released binaries?

Roy-043, 2024-11-13:
I am using this release binary:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/1.0rc3/FreeCAD_1.0.0RC3-conda-Windows-x86_64-py311.7z

That needs to updated then.

PaddleStroke, 2024-11-13:
I can confirm there is a problem in RC3. The backport of https://github.com/FreeCAD/FreeCAD/pull/17686 has failed.

Looking at FreeCAD-1.0 branch on github, the changes to TaskAssemblyCreateJoint.ui are missing. There is only the changes to JointObject.py. And so the error you have.

So main branch is not broken. Only FreeCAD-1.0.

maxwxyz, 2024-11-13:
@chennes FYI

adrianinsaval, 2024-11-13:
rc4 incoming

closed 2024-11-13 by adrianinsaval