FreeCAD/FreeCAD #17834: Part: ViewObject.Selectable is broken for App::Links

PaddleStroke, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Create new document
- Add a Part::Box
- Add a App::Link of the box
- Change Box 'Selectable' to false
- You can confirm that Link 'Selectable' property is still to True.

Expected : Box cannot be selected but Link can be selected

Observed : Box and Link both cannot be selected.

Closely related to https://github.com/FreeCAD/FreeCAD/issues/14219

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NomAnor, 2025-03-18:
This is still present in the main branch. According to my tests:

1. Box `Selectable = true`; Link `Selectable = true`: Both are selectable
2. Box `Selectable = true`; Link `Selectable = false`: Box selectable, Link not
3. Box `Selectable = false`; Link `Selectable = true`: Both not selectable
4. Box `Selectable = false`; Link `Selectable = false`: Both not selectable

The bug is in 3., if the box is not selectable it seems to override the selectable setting of the link.

```
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40845 (Git)
Build date: 2025/03/18 14:59:22
Build type: Debug
Branch: main
Hash: c1ada6bc4070df7804ccea69773b0cfa0d8226fd
Python 3.13.2, Qt 5.15.16, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * fasteners 0.5.34
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * 3DfindIT 1.2.0
``

NomAnor, 2025-03-19:
I tried to investigate more. Looking at the scene inspector, it looks like the selection node of the linked object is reused (`Cube`, `Box(0)`):

[image]

When `Selectable` of the linked object is set to false `PickStyle` is set to unpickable. Because the node is reused for the link, the pick style also applies to the link. It seems the pick style is checked before calling `isSelectable()` and so the link can't be selected even if `isSelectable()` would return true.

When `Selectable` of the linked object is set to true, no `PickStyle` is present and the result of `isSelectable()` for the link determines if it is selectable.

I have no clue how the scene graph works so I could not find a solution.