FreeCAD/FreeCAD #17835: Core: Dependency Graph: Graphviz failure: Graphviz failed to create an image file

AIRCAP, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Steps to reproduce:**
1. Unzip and open Fail.FCStd
[Fail.zip](https://github.com/user-attachments/files/17731459/Fail.zip)
2. Click on Tools->Dependency Graph

**Expected Outcome:**
Dependency Graph displayed

**Actual Outcome**
Popup: "Graphviz failure: Graphviz failed to create an image.
[image]

**Workaround**
Exporting the dependency graph in graphviz format works. Theres only a few warnings:
Tools->Export Dependency Graph ---> test.gv
```bash
dot -T png test.gv -o test.png
Warning: node '49', graph '' size too small for label
Warning: node '25', graph '' size too small for label
Warning: node '29', graph '' size too small for label
Warning: node '11', graph '' size too small for label
Warning: node '47', graph '' size too small for label
Warning: node '37', graph '' size too small for label
Warning: node '53', graph '' size too small for label
Warning: node '48', graph '' size too small for label
Warning: node '30', graph '' size too small for label
Warning: node '61', graph '' size too small for label
Warning: node '50', graph '' size too small for label
Warning: node '3', graph '' size too small for label
Warning: node '28', graph '' size too small for label
Warning: node '6', graph '' size too small for label
Warning: node '41', graph '' size too small for label
Warning: node '10', graph '' size too small for label
Warning: node '12', graph '' size too small for label
```
[image]

### Full version info

```shell
Latest main:

OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39188 (Git)
Build type: Debug
Branch: main
Hash: 4bc8cd60e9a9c24ed007f0d3d341ca70498c798a
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Defeaturing 1.2.2
  * Silk 0.1.5
  * CurvedShapes 1.0.12
  * CfdOF 1.27.13
  * Curves 0.6.51
  * Plot 2024.10.23
  * AirPlaneDesign 0.4.1
  * lattice2 1.0.0
```

### Subproject(s) affected?

Core

### Anything else?

Other files work fine, its just this particular dependency graph that fails. If "Line1" is deleted, Graph display works.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-11-13:
I ran into this when testing this new macro:
https://github.com/FreeCAD/FreeCAD-macros/pull/180

maxwxyz, 2024-11-13:
Could this be related to the recently merged https://github.com/FreeCAD/FreeCAD/pull/16427 ? @hasecilu

adrianinsaval, 2024-11-13:
does the appimage work?

hasecilu, 2024-11-13:
Works on the weekly build 39176 from 18 hours ago, Arch Linux Wayland.

running `dot -Tsvg -Granksep=2 -Goutputorder=edgesfirst -Gsplines=ortho file.gv -o file.svg` works?

If you say it's only that file it's not related to recent changes, I gues you have self compiled version, could you try the AppImage?

Your file

[image]

Another file

[image]

AIRCAP, 2024-11-18:
I managed to get a test run of the ApImage and I tested RC4 and the latest weekly. Error doesn't happen there, only in self compiled main. Any idea what could cause that? Is FreeCAD using some weird graphviz python library that could be different from the system wide dot?

adrianinsaval, 2024-11-18:
> Is FreeCAD using some weird graphviz python library

noup, but the appimage includes it's own version of graphviz. Are both dot and unflatten available in your path?

AIRCAP, 2024-11-20:
> > Is FreeCAD using some weird graphviz python library
> 
> noup, but the appimage includes it's own version of graphviz. Are both dot and unflatten available in your path?

Yes. As marked before, it the graph generation works fine with virtually all other FCStd files I have - including this one if that BSpline called "Line" is removed.

```shell
$ which unflatten
/usr/bin/unflatten
$ which dot
/usr/bin/dot
$ dot -V
dot - graphviz version 2.43.0 (0)
```

unflatten doesn't have a `-V` flag for version, but since its bundled I assume its the same

hasecilu, 2024-11-20:
I also have a self-compiled Qt6 version and also worked there, not sure what have could be wrong, if it's no causing troubles on other files and can't be reproduce will be hard to follow.

dzid26, 2025-01-07:
Dependency Graph never worked for me in various versions of Freecad.
I am not sure if height/width is a problem, since the dot works with it anyway

```
digraph "" {
graph [
compound=true];
subgraph clusterBody {
graph [
bgcolor="#69e2c550", label=Body, style="rounded,filled"];
subgraph clusterOrigin {
graph [
bgcolor="#15ba7050", label=Origin, style="rounded,filled"];
}
subgraph clusterOrigin {
graph [
bgcolor=none, label=Origin];
1[label="YZ_Plane&#92;n(YZ-plane)", shape=Mrecord, style=filled];
2[label="XY_Plane&#92;n(XY-plane)", shape=Mrecord, style=filled];
3[fixedsize=true, height=0, style=invis, width=0];
4[label="X_Axis&#92;n(X-axis)", shape=Mrecord, style=filled];
5[label="Y_Axis&#92;n(Y-axis)", shape=Mrecord, style=filled];
6[label="Z_Axis&#92;n(Z-axis)", shape=Mrecord, style=filled];
7[label="XZ_Plane&#92;n(XZ-plane)", shape=Mrecord, style=filled];
8[label=Origin001, shape=Mrecord, style=filled];
}
0[fixedsize=true, height=0, style=invis, width=0];
9[label=Sketch, shape=Mrecord, style=filled];
10[label=Pad, shape=Mrecord, style=filled];
9 -> 2;
10 -> 9[label=" 2x"];
}
}

```

`dot -Tsvg -Granksep=2 -Goutputorder=edgesfirst -Gsplines=ortho cube.gv -o cube.svg`
`Warning: node '3', graph '' size too small for label`
`Warning: node '0', graph '' size too small for label`
`Warning: Orthogonal edges do not currently handle edge labels. Try using xlabels.`
[image]

```
 dot -v
dot - graphviz version 2.50.0 (20211204.2007)
libdir = "C:\Program Files\Graphviz\bin"
Activated plugin library: gvplugin_dot_layout.dll
Using layout: dot:dot_layout
Activated plugin library: gvplugin_core.dll
Using render: dot:core
Using device: dot:dot:core
The plugin configuration file:
        C:\Program Files\Graphviz\bin\config6
                was successfully loaded.
    render      :  cairo dot dot_json fig gdiplus json json0 map mp pic pov ps svg tk vml xdot xdot_json
    layout      :  circo dot fdp neato nop nop1 nop2 osage patchwork sfdp twopi
    textlayout  :  textlayout
    device      :  bmp canon cmap cmapx cmapx_np dot dot_json emf emfplus eps fig gif gv imap imap_np ismap jpe jpeg jpg json json0 metafile mp pdf pic plain plain-ext png pov ps ps2 svg tif tiff tk vml xdot xdot1.2 xdot1.4 xdot_json
    loadimage   :  (lib) bmp eps gif jpe jpeg jpg png ps svg
```

leoheck, 2025-09-15:
I am still seeing this issue.

The generated file has fixedsize `with =0`  maybe the decimal places were cut from the out put.

[image]

Then this simple command would fail.

```
/tmp/fc 
✦ ➜ dot -Tsvg lala.gv -o lala.svg
Warning: node '2', graph '' size too small for label
Warning: node '5', graph '' size too small for label
```

A quick fix to update the hardcoded font size fixes the file.
[image]

[image]

[image]

```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/12 03:41:38
Build type: Release
Branch: main
Hash: 952bff5b63a62b55a7ea1623edfb39e90dee2d4c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
 ```

closed 2026-09-03 by Lgt2x