FreeCAD/FreeCAD #17836: [Tree] Regression in RC3 and main, drag and drop of object and it's mirror into Part container is nolonger allowed

Syres916, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If an object and it's mirror are both selected, previously they could be dragged and dropped onto a Part Container, this is nolonger allowed but IMHO cannot see a legitimate reason as to why this should be the case.

The last working build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

First non-working build I have shown below.

See https://forum.freecad.org/viewtopic.php?t=92070

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * Assembly4 0.50.15
  * A2plus 0.4.68
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-13:
@PaddleStroke could you take a look at this?

PaddleStroke, 2024-11-18:
I don't understand, why are you trying to move both the mirror and its children item? Just drag the symmetry object and it works correctly.

Syres916, 2024-11-18:
Because if I've got a 9 level deep tree and I only want to take the top 3 levels then as a user I should have that option, you're dictating what I can/cannot do which I could previously. Please advise where the DWG stated this was _progress_ in the UI, for me (yet) another reason for not upgrading to 1.x.

PaddleStroke, 2024-11-18:
Can you please stop aggressive tone, I just ask a question to understand the issue. I'm not dictacting anything. This is a side effect of a PR not a conscious decision to piss you off............................

I still don't understand what you mean. If you have 9 level deep and want only the top 3, first top in what direction? Please provide a clear example and repeatable steps. Thanks

closed 2025-09-13 by github-actions[bot]