FreeCAD/FreeCAD #17839: Importing step file in 1.0 RC2 will not work.

Herrcookiekiller, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When importing a step file that has been _exported from a project with FreeCAD 1.0 RC2_, it does not work.

•Importing the step file into 1.0 RC2, where it previously has been exported, only gives a single error.
•Importing the same step file into 0.21.1 works.
•Exporting as .step from 0.21.1 with the same export settings as 1.0 RC2, then importing that file into 1.0 RC2 works
(i set both FreeCAD versions to the same export and import settings)

Here is the step file causing the problem:
[mounting-hole-covers-injection-set.zip](https://github.com/user-attachments/files/17733468/mounting-hole-covers-injection-set.zip)

This is the error that FreeCAD 1.0 RC2 throws out
```
14:49:37  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\freecad\module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```

FreeCAD 0.21.1
[image]

FreeCAD 1.0 RC2
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.65
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * InventorLoader 1.4.0
  * kicadStepUpMod 11.2.4
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * OpticsWorkbench 1.0.18
  * sheetmetal 0.4.22
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-11-13:
I think this might have already been fixed with https://github.com/FreeCAD/FreeCAD/pull/16847

@Herrcookiekiller please test 1.0rc3 or the latest weekly build

adrianinsaval, 2024-11-13:
#16859 might be related too

maxwxyz, 2024-11-13:
Cannot reproduce on 
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39175 (Git) Conda
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5
```

Please test on the latest RC3 or weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Herrcookiekiller, 2024-11-13:
I just tried reproducing this on my PC at home, same FreeCAD 1.0 RC2 version but Windows 10.
Works just as fine as 0.21.1

I suspect this to be Windows 11 related.

adrianinsaval, 2024-11-13:
reproduction is likely not related to the file itself but to the path of the file. If I'm correct this bug present itself when there are non latin characters in the file path. Does the full path to the file in your failing machine contain any non standard character?

Herrcookiekiller, 2024-11-14:
> reproduction is likely not related to the file itself but to the path of the file. If I'm correct this bug present itself when there are non latin characters in the file path. Does the full path to the file in your failing machine contain any non standard character?

Nothing out the ordinary, also other step files work just fine from within the same folder.

Herrcookiekiller, 2024-11-14:
> Cannot reproduce on
>
> ```
> OS: Windows 11 build 26100
> Architecture: x86_64
> Version: 1.1.0dev.39175 (Git) Conda
> Build type: Release
> Branch: main
> Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
> Python [...]

I couldnt find 1.0 RC3, tried it on the weekly build.
This works fine, but i suspect it only works fine because it runs from within the folder and doesnt have to install anything for "All Users" or "Just Me" like it has to when running the installer for 1.0 RC2.

wwmayer, 2024-11-14:
`except PyExc_FC_AbortIOException:`
This is part of the error message and this class has never existed. So, you definitely used a version that suffered from this problem.

Herrcookiekiller, 2024-11-14:
> `except PyExc_FC_AbortIOException:` This is part of the error message and this class has never existed. So, you definitely used a version that suffered from this problem.

I downloaded from here: https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2
This file: `FreeCAD_1.0.0RC2-conda-Windows-x86_64-installer-1.exe `

And installed "For all users" under Windows. Why this would be bringing up this error seems so weird to me.

Anyways, RC4 just dropped as i saw, im sure everything will work there and this is just an extreme edge case with RC2 and my specific Windows 11 installation.

closed 2024-11-14 by Herrcookiekiller