FreeCAD/FreeCAD #17841: Draft: Add Group editing options to the Project Tree RMB context menu

semhustej, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In BIM WB, there are useful tools available from the RMB Context Menu when right-clicking on an object in the project tree. These would be very useful to have in Draft WB too (I would argue some of them would be useful in all workbenches). These tools can be found in other places in the WB, but having them in the RMB context menu is the most logical and convenient in my opinion. Having them here would greatly enhance user experience.

These tools are:

- Select group
- Add to construction group
- Move to group....
- Reorder children

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc3)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-13:
Interesting, but for example "Move to Group" is not in the "Part" workbench)
Just for your information. Doesn't directly concern this discussion.

Heh... someone will have to clean up sometime.

semhustej, 2024-11-14:
> Interesting, but for example "Move to Group" is not in the "Part" workbench)
> 
> Heh... someone will have to clean up and do an audit someday.

I think most of these options would be useful in all workbenches, but I do not have much experience with Part or Part Design modelling, so I can not propose that in a good consciousness.

semhustej, 2025-06-20:
Solved partially (selection of group contents) by https://github.com/FreeCAD/FreeCAD/pull/22082