FreeCAD/FreeCAD #17844: Draft: Layer changes object properties unexpectedly when it is interacted with

semhustej, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In certain workflows, layer changes properties of it's content unexpectedly.

Steps to reproduce:

- open file 
[layer-behavior.FCStd.zip](https://github.com/user-attachments/files/17737914/layer-behavior.FCStd.zip)

- move the Rectangle to the New Layer -> Rectangle changes color (and other view properties) according to layer properties to green 
- change the color of the Rectangle in it's properties to red
- create new objects, edit them, do other stuff. The Rectangle is still red
- move another object to the New Layer
- The Rectangle is suddenly Green

See behavior in video:

https://github.com/user-attachments/assets/738ed3fb-21bc-4c18-b745-db7b59e9f80a

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc3)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

I am not sure, what is desired behavior by FreeCAD
a) Properties of objects inside of layers can be changed independently of the layer properties. In that case the property of the object should change back to layer property only if requested by the user.
b) Properties of objects inside of layers can not be changed independently of the layer properties. In that case view properties influenced by the layer should be locked for editing in property of object inside of a layer.

The current behavior is something in between and is causing confusion in my opinion.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@Roy-043 FYI

yorikvanhavre, 2024-11-25:
This is a complex problem. The layer contains code to change the color of its children. Now if a child changes its color afterwards, there is no way for the layer to know it. And indeed on next run of the layer code, it will "re-paint" all of its children.

I don't see an immediate solution for this... Of course we could add a callback on all Draft or BIM objects to check for their layer when changing their color, but that won't work for all FreeCAD objects, so it's not optimal

Tying the color to the layer color via expressions I think is not yet supported, and in any case would cause a circular dependency, so wou;dn ´t be doable either.

To be honest I have no good idea here :(

semhustej, 2024-11-28:
From what I understand:

1. would mean, that when a new object is placed in the layer, only this new object will receive layer properties. That would mean, that user can change property of an individual object inside of the layer and this property would stay this way even if user places other objects in the layer. 
_I think this is desirable behavior._

2. would mean that if user changes property of an individual object O inside of the layer and than changes the layer property, the changed property of an individual object O would stay as it was set by the user - independent of the layer property.
_I think this behavior is more clear for the user to understand, than the current behavior._

Do I understand this correctly?
If a user changes individual object O property and than changes his/her mind and wants the object to acquire the property from the layer setting, it could be achieved by setting the property the same as the layer property. According to 2., if Layer property is than changed, FC would detect similar value in both of these and change the property of an object O according to changes in layer property.

I am heavily influenced by using standard DWG editing software for years and these changes would make Layer behavior closer to those, so these sound good to me. I think the proposed behavior would be more intuitive for users even without previous DWG editor experience.

Roy-043, 2025-01-09:
I am attaching a file that I think provides a reasonable solution:
1. When an object is put in a layer it always adopts the properties of that layer.
2. Properties of the object that are subsequently changed are considered overrides.
3. When the properties of the layer, or the object in the layer, are then changed so that they match again, there is no longer an override, and properties are synced again.
4. The layer and the layer container object get an extra Treeview context menu option "Apply properties of layer to contents"/"Apply properties of all layers to contents".

[view_layer.py.txt](https://github.com/user-attachments/files/18360775/view_layer.py.txt) (remove the .txt extension)

semhustej, 2025-01-10:
@Roy-043 I tested the library that you linked in the previous post. I did only a simple file. These are my findings:

- **Should be fixed if possible:**
The original post issue is still present. I tried with the original test file a followed steps from OP.
Proposed solution: If possible, moving object to a layer should influence properties of only that one object. Not objects which have overridden properties.

- Apply properties to all layer contents – works well. This is a nice solution.

- Apply properties of layer to contents  – works. This is a nice solution, even though a bit hidden.

- The situation with overridden object properties work as you describe and I think is nice and understandable for user.

- **Should be fixed if possible:**
The situation with overridden object properties does not work with line color in my testing. When Layer property Line color is change, even objects, which do not have the property overridden (those that have the same line color as the layer) do not change line color according to the new setting. Only point color is changed. This is the only property that does not work properly I think. I tested others (Line Width, Draw Style, Face Color).
See video:

https://github.com/user-attachments/assets/b53fcc99-bac2-4ed1-b41e-0e96535a46be

As far as I know Line Print Color is not possible to change at individual object level so I did not test this.

semhustej, 2025-01-10:
I also encountered an issue with changing transparency in Draft_LayerManager to 0.

Steps to reproduce:

- open new file
- open Draft Layer Manager
- create a new layer
- change transparency of the new layer to 80
- close layer manager with `OK`
- open layer manager
- change the transparency of the New Layer to 0
- close layer manager with `OK`
- the transparency of the layer is not changed, is still 80

I am able to reproduce this with FC 1.0, so I think it has nothing to do with the changes that you made. Let me please know, whether I should report this as a separate issue. I could also reproduce the issue with BIM_layers tool.

See video:

https://github.com/user-attachments/assets/3a28950f-048e-422c-90d8-d797899e4d24

Roy-043, 2025-01-11:
PR for the transparency issue: #18988.

Roy-043, 2025-01-11:
Improved code for this issue.
[view_layer.py.txt](https://github.com/user-attachments/files/18383466/view_layer.py.txt) (remove the .txt extension)

semhustej, 2025-01-11:
- the OP issue is fixed with the latest library. **Fixed.**
- line color changes for all non-overridden objects. **Fixed.**

One little thing (not a deal breaker) - I am not able to synchronize Face color by selecting the same color as the layer color. I tried changing both `Shape appearance` property and `Diffuse color` property. None of them seem to synchronize overridden object to layer, such as for instance Line color does.

See the behavior in video:

https://github.com/user-attachments/assets/ad7f9f21-1085-4174-847e-ae93e115f964

Roy-043, 2025-01-12:
> I tried changing both `Shape appearance` property and `Diffuse color` property. None of them seem to synchronize overridden object to layer, such as for instance Line color does.

When the DiffuseColor is changed a new ShapeAppearance is created. Unexpectedly some rounding is applied, affecting not just the edited color, but all colors of the ShapeAppearance. In de video @ 0:15/0:16 you can see that the AmbientColor also changes.

Attached new code that takes this into account (and also ignores the alpha component of the colors).
[view_layer.py.txt](https://github.com/user-attachments/files/18390833/view_layer.py.txt)

semhustej, 2025-01-13:
That is a good catch. The last version works around this issue. **From my testing it seems everything with the new library is OK.**

But is this a good practice (regarding the last issue)? Shouldn't the underlaying color mismatch be handled, even though I guess it is not a Draft issue, or is that too complicated?

I am able to reproduce mismatches in set colors in all pre-set colors. That means, that what I am choosing is different than the result. Plus there is a mismatch of information between two places that give me this information in FreeCAD:

[image]

I do not see this as a problem in my practice, but I guess earlier or later someone will run into one.

Roy-043, 2025-01-13:
Issue for the rounding of color components: #19048.

PR #19207 (2025-01-28): Draft: allow objects in layers to have overrides
Fixes #17844.

1. When an object is put in a layer it always adopts the properties of that layer.
2. Properties of the object that are subsequently changed are considered overrides.
3. When the properties of the layer, or the object in the layer, are then changed so that they match again, there is no longer an override, and properties are synced again.
4. The layer and the layer container object get an extra Tree view context menu option "Reassign properties of layer"/"Reassign properties of all layers".
commit dfe670e1 (2025-01-28): Draft: allow objects in layers to have overrides (#19207)

* Draft: allow objects in layers to have overrides

Fixes #17844.

1. When an object is put in a layer it always adopts the properties of that layer.
2. Properties of the object that are subsequently changed are considered overrides.
3. When the properties of the layer, or the object in the layer, are then changed so that they match again, there is no longer an override, and properties are synced again.
4. The layer and the layer container object get an extra Tree view context menu option "Reassign properties of layer"/"Reassign proper [... cut]
closed 2025-01-28 by Roy-043