FreeCAD/FreeCAD #17845: radius on curved shape problem

Roberto8303, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when I try to execute a radius on the curved shape of the solid, two unwanted bodies are formed. I attach a photo
[image]
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19042
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Behave-Dark-Colors 0.1.1
  * CamScripts
  * ExplodedAssembly
  * fasteners 0.5.29
  * FeedsAndSpeeds 0.5.0
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
  * PieMenu 1.8.0
  * ProDarkThemePreferencePack 1.0.0
  * QuickMeasure 2022.10.28
  * Render 2024.9.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roberto8303, 2024-11-13:
[raggio.zip](https://github.com/user-attachments/files/17738799/raggio.zip)

pierreporte, 2024-11-13:
I can confirm.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc3)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * sheetmetal 0.5.4
```

I thought that the problem was that the datum plane, thus the sketch, is coincident with the edge common with the planar and cylindrical surface, because it works when applying an offset. The model looks fine though it is extremely slow to load, and deleting the fillet took a lot of time during which FreeCAD was frozen. However, I managed to make a similar fillet in a part design from scratch that can be downloaded here: [radius bug.zip](https://github.com/user-attachments/files/17739070/radius.bug.zip)

[image]

[image]

maxwxyz, 2024-11-14:
Does this happen to be an already known issue: https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22Topic%3A+Fillets%22
Is it the same in the stable 0.21.2 version?

Roberto8303, 2024-11-14:
**IOr I solved it by moving the reference plane in z by 0.01mm so as to separate the two bodies. This way it works. I don't know if this information can be useful to try to solve the problem**

pierreporte, 2024-11-14:
@Roberto8303 It’s not a good solution though. A better approach would be to make to different bodies, then join them using a boolean operation. The problem is real though it could just be a deeper problem with your file.