FreeCAD/FreeCAD #17846: PartDesign: Automatic refine cause issues with Helixes

FlachyJoe, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

#14406 has set automatic refine the default behavior but refinement can produce not good enough shape in some cases in peculiar if it's used as Base for a helix operation.
ref: #17829 #17730 #16081

When the user actively switch on the Refine property and see its model fail he can undo but if the refine is always on he can not know the cause of the failure.

### Full version info

```shell
since 0.21.37966
```

### Subproject(s) affected?

PartDesign

### Anything else?

I see two possibility: 
- remove the auto refinement
- find a better check for shape validity and auto switch off Refine if the shape is worse. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
Is there a chance that the helix feature ignores the refine property as workaround?

FlachyJoe, 2024-11-14:
@maxwxyz if refine is applied on the previous shape then the helix feature can't change it. The main cause is in fact the refine process: the produced shape is not always as clean as the not-refined one.

FlachyJoe, 2025-02-05:
Refine causes issue out of hellX too: #19410

FlachyJoe, 2025-10-26:
@FEA-eng yes, this report covers different issues with the same cause. It needs to work on the refinement algorithm or on a cleaning of the produced shape.
The just merged #23486 PR prevents a refinement failure to restrain the user to work but this current problem appear with a working refine so it is not touched.