FreeCAD/FreeCAD #17848: wrong feed speed being generated

jeffgrain, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

wrong feed speed seems to be generated during the post process.. did a simple test drawing , entered feeds and speeds in the tool setup.. generated the GRBL g-code.. then opened the file in a text editor to check.. it looks to me like the feed speed is being set to F21000 starting at line 26 .. if I understand it correctly

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]
![test part](https://github.com/user-attachments/assets
[image]
/fa700267-78c0-4af7-88c6-b48a0d73136d)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
Is this the same issue in the latest RC4 and also as in the stable 0.21.2 version? Can you share an example file?

jeffgrain, 2024-11-14:
Sorry to say but I have just started working with the CAM workbench..  and since the new version is coming out decided to migrate over.. so no.. I have not tried it yet.. If I have time later I will  try.. here are two examples.. a part I made and a test part I did to check into this problem.. you will notice that the feed speed on one if f6000 and the test one is f21000.. I would be happy to share.. but I can not seem to attach the files.. if you wish to send me a reply email.. to jeffgrain@gmail.com  I will gladly send them along``

jeffgrain, 2024-11-14:
[files.zip](https://github.com/user-attachments/files/17754417/files.zip)

ok try this

jeffgrain, 2024-11-14:
ok tried same test on freecad 0.21.2 linux os.. same result.. repeated on windows bases 1.0 rc2 and it appeared to work.. the correct feed speeds where in the gcode file..
hope this helps.

maxwxyz, 2024-11-14:
then it is already fixed in 1.0

closed 2024-11-14 by maxwxyz

jbaehr, 2024-11-19:
There is no fix in 1.0, as there was no error in the first place. In your screenshot you configure a horizontal feed rate of 350 *mm/s*, and your gcode shows F21000, which is in *mm/min*. And 350 * 60 is 21000 so this looks good to me.

On the windows machine, where you see your expected feed rates, you've most likely used the unit scheme "Metric Small Parts & CNC", as recommended in the [CAM WB Docs](https://wiki.freecad.org/CAM_Workbench#Units) and a configured feed rate of 350 *mm/min*,

jeffgrain, 2024-11-19:
yes you are right.. the error was in not having the correct mm/s to mm/min ,, it did not get set to mm/min I am uncertain of why.. it was changed and not saved for some reason.. it is set properly now.. and all looks very good.. there was one minor "glitch" just after I changed it .. but I have not been able to reproduce it.. 

So far with 1.0 release,, the work flow is going smoothly.. and is looking very professional.. 

now for the real test.. putting it to work on a machine..  thanks much