FreeCAD/FreeCAD #17849: FreeCAD Crash on Mac OS Sequoia

wozzan8tor, 2024-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Mouse click in the file tree Origin design planes causes FreeCAD to crash instantly. I recall seeing another report of this, again on a Mac. See full description for time stamped bug reports.

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default

I have a project saved with a closed sketch and after a successful 'pad'. I subsequently re-opened this project and clicked in the file tree Origin design planes causing FreeCAD to crash instantly.
Extracts from Apple generated logs.
{"timestamp":"2024-11-10 09:44:46.20 +1300","bug_type":"226","os_version":"macOS 15.0.1 (24A348)"}
A second attempt clicking in another plane of Origin:
{"timestamp":"2024-11-10 09:44:46.37 +1300","bug_type":"226","os_version":"macOS 15.0.1 (24A348)"}
```

### Subproject(s) affected?

None

### Anything else?

Sorry, maximum is 65536 characters here, Apple crash report is 146KB when zip'd.

And Screen capture is 205 KB when zip'd.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
Should be fixed in latest RC4 or weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Please comment if it is still an issue.

closed 2024-11-14 by maxwxyz

wozzan8tor, 2024-11-14:
Hello Max, 
Thank you for the quick response, not sure if you intend comments to take the form of a reply…two day old noob GitHub user, two week old FreeCAD experimenter. 

Grabbed the latest build arm64 dmg, this is the result -

￼

maxwxyz, 2024-11-14:
There is no result displayed, can you comment directly on GitHub? You can paste images, video or text and also .zip files.