FreeCAD/FreeCAD #17850: RC2 expression editor: can multiply int or float, cannot add either

mpictor, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a sketch, I have a dimension that I've named `sh_maj`. I created another dimension and opened the expression editor, typing in `Constraints.sh_maj + 3`. I got an error, so tried float rather than int - same result.

If I replace `+` with `*`, it works with int and float.

I'd expect any value that works with `*` to also work with `+`.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Lithophane
  * OpticsWorkbench 1.0.9
```

### Subproject(s) affected?

None

### Anything else?

Addition fails with int or float
[image]

Multiply works with int or float
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-14:
Try + 1.8 mm instead of + 1.8

These are unit quantity variables and you're trying to add a unitless value to them.

mpictor, 2024-11-14:
Interesting, that works.

I am fairly confident that previous FreeCAD versions assumed `mm` if no unit was given, so this kinda seems like a regression.

I did manage to gloss over the fact that the error includes `unit mismatch` - I guess in my head I translated that to `type mismatch`.

WDYT about an error message like this?
```diff
-Quantity::operator+(): Unit mismatch in plus operation in expression ...
+Incompatible units - left: mm, right: <none>
```

Taking it a step further (though I imagine it'd be considerably more work):
Have two distinct colors for the right and left operands, and color the units in the error the same way.

mwganson, 2024-11-14:
Not a regression as far as I know.  I suppose an assumption could be made that the default unit system is intended, but I think it's better to error out and force the user to specify the unit type rather than to guess and get it wrong.

As for the error message no doubt there is some room for improvement.  Depending on the context of where the message is being created it could difficult to be more precise than simply telling the user there is a unit mismatch error in the expression..

maxwxyz, 2024-11-14:
So no bug but rather a feature request?

adrianinsaval, 2024-11-15:
> I am fairly confident that previous FreeCAD versions assumed mm if no unit was given, so this kinda seems like a regression.

not for this expression, if both operands are unitless the result is unitless and then the overall result is assumed to be whatever the base unit for the property is.

This is correct behavior, only the messaging could be improved, I like your suggestion of showing what units each operand has.

> I suppose an assumption could be made that the default unit system is intended, but I think it's better to error out and force the user to specify the unit type rather than to guess and get it wrong.

This kind of assumptions are dangerous and should NOT be made. On the other hand one could say that we do make them whit unitless results, which has been problematic for many imperial users that expect them to be interpreted as imperial but are interpreted as mm, so it might also be a good idea to display a warning for unitless expressions.

mpictor, 2024-11-16:
> not for this expression, if both operands are unitless the result is unitless and then the overall result is assumed to be whatever the base unit for the property is.

Ah, maybe the case where it did work and was interpreted as using the default unit was adding a spreadsheet value (unitless) and an int (also unitless).

NSUBB, 2025-02-14:
Hello, I would like to request that the units be inferred when the other operand has one defined. One argument is that this inference can cause unexpected results, but the expression dialog box provides the result. So the user should notice if the resulting value is different than they expect. Furthermore, multiplication will continue because it needs no unit, so I think it is natural for the user to expect the same behavior during addition or subtraction. That is, to not be required to declare a unit unless it differs from the other operand. Thank you for considering.

HostXnine, 2025-03-06:
In my case the problem is that I have a spreadsheet where I've defined all values in mm. The model is defined in mm as well, but when I use the expression editor I still get and unit miss match error. Although I've defined all values in the spreadsheet in mm the expression builder considers them as unitless and constrains as values in mm. I think that in my case this might be eligible for a bug because this is not an expected behaviour.

[image]
[image]
[image]

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/3f8dbf8040b542b902a136e7fd53c50b64c34485

The addition and subtraction error shows both units, for example left: mm, right: unitless. It still requires compatible units.