FreeCAD/FreeCAD #17853: FreeCAD 1.0rc3 wont open on macOS

Orion98MC, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi there, the RC3 releases for macOS (both x86 and ARM) would not open on my mac. macOS says the app is "damaged" and that the image should be ejected. I'm on a M1 mac and I tried both versions x86 and ARM.

### Full version info

```shell
FreeCAD 1.0rc3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-14:
RC3 had critical build error.. Stay tuned for RC4

Fran6nd, 2024-11-14:
Did you try to run 
```xattr -c /Applications/FreeCAD.app```
after installing it?
I believe we have an issue with how it is signed. No more troubles since I'm doin it.

Orion98MC, 2024-11-14:
@Fran6nd This indeed solved the launching issue! Thanks.
@luzpaz i'm looking forward for RC4 ;) Keep up ! I use FreeCAD a lot more now.

maxwxyz, 2024-11-14:
Closing this as RC4 is out.

closed 2024-11-14 by maxwxyz