FreeCAD/FreeCAD #17856: Start screen nearly unreadable with FreeCAD classic theme in RC4

berberic2, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If FreeCAD is startet the first time and the user selects the classic theme, the start-screen is broken.
The captions of new projects and examples is nearly unreadable, because it is white on light-gray.
(This might depend on dark/light theme of the desktop environment)
[Start.webm](https://github.com/user-attachments/assets/7d508ff1-eec4-408f-8546-1b454af41a8d)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0RC4.38641 +463 (Git)
Build type: Unknown
Branch: (HEAD losgelöst bei 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@MisterMakerNL @hyarion FYI

MisterMakerNL, 2024-11-14:
No just Freecad. But if not, than it is probably what I think it is.

hyarion, 2024-11-14:
This is only be a problem on linux when using qt < 6.4. 

I don't remember why, but we assumed that this wasn't a problem on linux, so we specifically didn't add a workaround for this.

This check is responsible for enabling the workaround.
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/Gui/ThemeSelectorWidget.cpp#L80-L90
```c++
static bool shouldHideClassicTheme()
{
    // Classic on macOS and windows 11 with qt6(.4+?) doesn't work when system
    // is in dark mode and to make matter worse, on macOS there's a setting that
    // changes mode depending on time of day.
#if QT_VERSION >= QT_VERSION_CHECK(6, 4, 0) || defined(FC_OS_MACOSX) || defined(FC_OS_WIN32)
    return true;
#else
    return false;
#endif
}
```

@chennes what do you think, should we always do this workaround to auto select, including on linux? (it will be light as autodetect doesn't work on qt < 6.4)

hyarion, 2024-11-14:
@berberic2 just to be 100% sure, the issue doesn't happen when clicking the dark theme in the theme selector, right?

To test initial behavior again, you can use safe mode to temporarily revert to default settings (help menu -> restart in safe mode)

berberic2, 2024-11-15:
The intention of this bugreport was not to get the classic theme removed, but to get it fixed.

MisterMakerNL, 2024-11-15:
Correct solution would be here that the buttons use the correct OS stylesheet. 
The asy solution here is to also make the text of the buttons not follow the stylesheet. And tho set it too black.

hyarion, 2024-11-15:
@MisterMakerNL yes indeed, for this view. But if it is anything like on macos and windows when in dark mode, then there's a lot of other similar issues in various places.
Fixing these would be really good, but can it be done before 1.0? If not what can be done if anything?

Doing the same thing on linux as we do on macos and windows could be a safe, predictable option.

Don't get me wrong, I would love to see an UI which automatically adjust to the system dark/light mode

MisterMakerNL, 2024-11-15:
@hyarion This is how classic is suppose to work so I don't think there are a lot of other issues. But also there are a gazillion linux stylesheets not all of them will be bug-free.
But also classic has been the goto theme for many years for many devs.

MisterMakerNL, 2024-11-15:
I see this as a blocker @maxwxyz but it's not that hard too fix tho.

MisterMakerNL, 2025-05-21:
I don't use linux, but the lack of any more complains about this I would conclude this can be closed.

closed 2025-05-21 by hyarion