FreeCAD/FreeCAD #17857: Icon-files of the startup screen are not removed from /tmp

berberic2, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For the start-screen an lot of icons get generated in `/tmp` which are not removed when freecad exists. The files are png-images containing the start-screen _icons_ and named `file□□□□□□`.
Using FreeCAD some time can easily result in hundreds of these files in` /tmp.`

This is an old problem, that was solved once but recurred ans still exists in RC4. See issue #10221

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0RC4.38641 +463 (Git)
Build type: Unknown
Branch: (HEAD losgelöst bei 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@qewer33 FYI @wwmayer you've made the PR back then: https://github.com/FreeCAD/FreeCAD/pull/10951

berberic2, 2024-11-14:
> @qewer33 FYI @wwmayer you've made the PR back then: #10951

See: https://forum.freecad.org/viewtopic.php?t=88201#p772834

> Das ist ein Fehler. Mit der alten Startseite hatten wir genau dasselbe Problem bis einschließlich v0.22. Das Problem wurde erst im Oktober mit [git commit 2bd89e5771](http://github.com/FreeCAD/FreeCAD/commit/2bd89e5771) behoben. Mit der neuen Startseite wurde das Problem leider wieder reaktiviert.

commit 2819ca43 (2024-11-20): Start: fixes #17857: Icon-files of the startup screen are not removed from /tmp

It's basically a port of #10951 to the new start page implementation.

Note: Icon files are not removed but re-used instead.

The commit adds some new functions:

* getThumbnailsImage()
  Returns the name of the PNG inside a project file

* getThumbnailsName()
  Returns the directory name containing the image files

* getThumnailsParentDir()
  Returns the parent directory of the directory containing the image files

* getThumbnailsDir()
  Returns the path to the thumbnail directory. There is no need to always crea [... cut]
closed 2024-11-20 by chennes