FreeCAD/FreeCAD #17860: RC2 Appearance menu for body missing options

Tony-g80, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When right clicking on body in tree view and selecting appearance from the options the ability to specifically change a parts color is missing, it was present in 021.2 but is now missing in RC2. 

The only option now for a part body to be changed regarding color is right click on body part and selecting random color, having multiple bodies in different colors is something i use alot with multi part models and specifically certain colors are chosen to aid with visibility of parts.
[image]
[image]
[image]

other than that im mclovin it lol.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * AirPlaneDesign 0.4.0
  * Assembly4 0.50.4
  * CurvedShapes 1.0.5
  * Curves 0.6.15
  * ExplodedAssembly
  * fasteners 0.4.71
  * frame 0.1.1
  * freecad.gears 1.1.0
  * Gridfinity 0.5.0
  * LCInterlocking
  * Lithophane
  * MakerWorkbench 1.0.1
  * parts_library
  * sheetmetal 0.3.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
Intended. You find it under either `Custom appearance` or assign a (custom) material

maxwxyz, 2024-11-15:
@davesrocketshop just to double check, this is correct?