FreeCAD/FreeCAD #17861: UI/UX: RC4 no classic theme offered

semhustej, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening default installation of FreeCAD RC4, I get only Light and Dark theme offered. There is no Classic theme. The same happens in RC3.

I deleted config files before I installed RC4.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-14:
Because of this [commit](https://github.com/FreeCAD/FreeCAD/commit/7e72d641df8985ab18ff31a83896cccea979c473)

maxwxyz, 2024-11-14:
Intended behavior

closed 2024-11-14 by maxwxyz

semhustej, 2024-11-14:
Thank you for the explanation. I was confused by seeing Classic Theme being offered in another issue on Linux.