FreeCAD/FreeCAD #17862: BIM: Several issues when default BIMexample is converted to IFC

semhustej, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are several problems when converting default BIMexample to IFC.

Steps to reproduce:

- open FreeCAD, open BIMExample.FCstd
- convert document to strict IFC mode
- observe several issues on the "Upper floor"

1. Several walls missing
2. No openings in walls for doors
3. Cap on the red construction
4. See screenshots

FreeCAD model:
[image]

Model after switching to strict IFC mode:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@yorikvanhavre could you take a look at the BIM example?

yorikvanhavre, 2024-11-15:
Ooh good catch. Good opportunity to refine IFC export. I'll have a look at these ASAP.
Could you compare it with when you simply export the model to IFC (using File->Export instead of the lock button?) then we can identify which are export problems, and which are specific NativeIFC problems

semhustej, 2024-11-15:
> Ooh good catch. Good opportunity to refine IFC export. I'll have a look at these ASAP. Could you compare it with when you simply export the model to IFC (using File->Export instead of the lock button?) then we can identify which are export problems, and which are specific NativeIFC problems

I tested with exporting to IFC ( File-> Export ). None of the highlighted issues occur in the exported IFC file (I checked by opening the IFC file freecad and also in BonsaiBIM).

So it looks like there's something going on during conversion to strict IFC mode.

yorikvanhavre, 2024-11-15:
Yes. The good part is, our export code is good :sweat_smile: 

Both systems use the same code. The main problem here is, the export tool takes a list of objects, and produces an IFC file. When using the lock button, the conversion is done object per object, and the relationships are created on the fly. That's where the problems arise.

Maybe we should change the strategy here, and use the export code to actually export an IFC file in one go, and simply reimport that file. That might be much more solid...

semhustej, 2025-01-19:
Hi, I tested with the latest daily and I am still missing walls when `IFC -> Expand children` is selected:

[image]

Additionally, the color of the round wall changes from red when `IFC -> Expand children` is executed.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

PR #18578 (2024-12-20): BIM: Several fixes to nativeIFC lock/unlock system
Fixes #17862

The main problem was actually just a stupid radians/degrees conversion...
Also fixed export units and other small details.

Introduces also an alternative conversion method that writes the IFC contents to a temp file then reloads it, but that methods ended up not useful anymore (it's also way slower) and is not used by current code. But I kept it as it can be useful for debugging at some point...
commit a7c550da (2025-01-06): BIM: Several fixes to nativeIFC lock/unlock system - fixes #17862