FreeCAD/FreeCAD #17863: BIM: Tutorial images are missing

maxwxyz, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum report: https://forum.freecad.org/viewtopic.php?t=92073
---
BIM ingame tutorial doesn't load images in 1.0 RC3, but it doesn't either in recent weekly builds:

[image]

1.I opened a BIM project
2.I click Help->BIM Tutorial
3.I have those messages: 

```
17:10:36  unparsable image path: /images/thumb/8/87/BIM_Tutorial_title.jpg/300px-BIM_Tutorial_title.jpg
17:10:36  unparsable image path: /images/thumb/0/01/BIM_Tutorial_01.jpg/300px-BIM_Tutorial_01.jpg
17:10:36  unparsable image path: /images/thumb/5/52/BIM_Tutorial_02.jpg/300px-BIM_Tutorial_02.jpg
17:10:36  unparsable image path: /images/thumb/e/ec/BIM_tutorial_14.jpg/300px-BIM_tutorial_14.jpg
17:10:36  unparsable image path: /images/8/82/BIM_tutorial_13.jpg
17:10:36  unparsable image path: /images/thumb/8/8c/BIM_tutorial_09.jpg/300px-BIM_tutorial_09.jpg
17:10:36  unparsable image path: /images/1/18/BIM_tutorial_10.jpg
17:10:36  unparsable image path: /images/thumb/5/56/BIM_Tutorial_03.jpg/300px-BIM_Tutorial_03.jpg
17:10:36  unparsable image path: /images/thumb/5/5a/BIM_Tutorial_04.jpg/300px-BIM_Tutorial_04.jpg
17:10:36  unparsable image path: /images/thumb/6/6b/BIM_Tutorial_05.jpg/300px-BIM_Tutorial_05.jpg
17:10:36  unparsable image path: /images/thumb/b/b6/BIM_tutorial_17.jpg/300px-BIM_tutorial_17.jpg
17:10:36  unparsable image path: /images/0/07/BIM_Tutorial_35.jpg
17:10:36  unparsable image path: /images/c/ce/BIM_Tutorial_33.jpg
17:10:36  unparsable image path: /images/7/73/BIM_Tutorial_34.jpg
17:10:36  unparsable image path: /images/thumb/5/57/BIM_Tutorial_06.jpg/300px-BIM_Tutorial_06.jpg
17:10:36  unparsable image path: /images/thumb/9/95/BIM_Tutorial_07.jpg/300px-BIM_Tutorial_07.jpg
17:10:36  unparsable image path: /images/6/6c/BIM_tutorial_15.jpg
17:10:36  unparsable image path: /images/thumb/a/a5/Arch_Wall.png/16px-Arch_Wall.png
17:10:36  unparsable image path: /images/3/33/BIM_tutorial_16.jpg
17:10:36  unparsable image path: /images/thumb/a/a5/Arch_Wall.png/16px-Arch_Wall.png
17:10:36  unparsable image path: /images/7/7b/BIM_tutorial_11.jpg
17:10:36  unparsable image path: /images/thumb/d/d1/BIM_tutorial_08.jpg/300px-BIM_tutorial_08.jpg
17:10:36  unparsable image path: /images/thumb/b/b6/BIM_tutorial_12.jpg/300px-BIM_tutorial_12.jpg
17:10:36  unparsable image path: /images/thumb/f/ff/BIM_Tutorial_20.jpg/300px-BIM_Tutorial_20.jpg
17:10:36  unparsable image path: /images/thumb/1/14/Draft_Rectangle.png/16px-Draft_Rectangle.png
17:10:36  unparsable image path: /images/thumb/3/30/BIM_Tutorial_18.jpg/300px-BIM_Tutorial_18.jpg
17:10:36  unparsable image path: /images/thumb/a/a4/BIM_Tutorial_19.jpg/300px-BIM_Tutorial_19.jpg
17:10:36  unparsable image path: /images/thumb/2/22/BIM_Slab.png/16px-BIM_Slab.png
17:10:36  unparsable image path: /images/thumb/0/01/BIM_Tutorial_21.jpg/300px-BIM_Tutorial_21.jpg
17:10:36  unparsable image path: /images/thumb/c/c5/Draft_Move.png/16px-Draft_Move.png
17:10:36  unparsable image path: /images/thumb/a/aa/Draft_SelectPlane.png/16px-Draft_SelectPlane.png
17:10:36  unparsable image path: /images/thumb/c/c5/Draft_Move.png/16px-Draft_Move.png
17:10:36  unparsable image path: /images/thumb/f/f5/BIM_Tutorial_23.jpg/300px-BIM_Tutorial_23.jpg
17:10:36  unparsable image path: /images/thumb/a/aa/Draft_SelectPlane.png/16px-Draft_SelectPlane.png
17:10:36  unparsable image path: /images/thumb/2/2c/BIM_Tutorial_22.jpg/300px-BIM_Tutorial_22.jpg
17:10:36  unparsable image path: /images/thumb/d/de/BIM_Tutorial_24.jpg/300px-BIM_Tutorial_24.jpg
17:10:36  unparsable image path: /images/thumb/0/05/BIM_Tutorial_25.jpg/300px-BIM_Tutorial_25.jpg
17:10:36  unparsable image path: /images/thumb/2/28/BIM_Tutorial_26.jpg/300px-BIM_Tutorial_26.jpg
17:10:36  unparsable image path: /images/thumb/d/d2/Arch_Add.png/16px-Arch_Add.png
17:10:36  unparsable image path: /images/thumb/f/f9/BIM_Tutorial_27.jpg/300px-BIM_Tutorial_27.jpg
17:10:36  unparsable image path: /images/thumb/e/e2/BIM_Tutorial_28.jpg/300px-BIM_Tutorial_28.jpg
17:10:36  unparsable image path: /images/thumb/2/23/BIM_Tutorial_29.jpg/300px-BIM_Tutorial_29.jpg
17:10:36  unparsable image path: /images/thumb/1/1a/BIM_Tutorial_30.jpg/300px-BIM_Tutorial_30.jpg
17:10:36  unparsable image path: /images/thumb/a/ab/BIM_Tutorial_31.jpg/300px-BIM_Tutorial_31.jpg
17:10:36  unparsable image path: /images/thumb/0/07/BIM_Tutorial_32.jpg/300px-BIM_Tutorial_32.jpg
17:10:36  unparsable image path: /images/7/72/BIM_Tutorial_36.jpg
17:10:36  unparsable image path: /images/c/c4/BIM_Tutorial_37.jpg
17:10:36  unparsable image path: /images/5/55/BIM_Tutorial_38.jpg
17:10:36  unparsable image path: /images/4/42/BIM_Tutorial_39.jpg
17:10:36  unparsable image path: /images/7/73/BIM_Tutorial_34.jpg
17:10:36  unparsable image path: /images/5/52/BIM_Tutorial_40.jpg
17:10:36  unparsable image path: /images/thumb/b/b0/Draft_Dimension.png/16px-Draft_Dimension.png
17:10:36  unparsable image path: /images/1/1a/BIM_Tutorial_41.jpg
17:10:36  unparsable image path: /images/thumb/9/9f/Draft_Text.png/16px-Draft_Text.png
17:10:36  unparsable image path: /images/5/5e/BIM_Tutorial_42.jpg
```

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39175 (Git) Conda
Build type: Release
Branch: main
Hash: 46831f4c86a1679d48f63273b92cd755522e42f9
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@yorikvanhavre should this be fixed and backported to 1.0?

yorikvanhavre, 2024-11-15:
I think it's late for 1.0... For translations unfortunately ATM there is no support for it, that's something we need to work on

PR #17878 (2024-11-18): BIM: Fixed image loading of the tutorial
fixes #17863
commit 36cf031f (2024-11-18): BIM: Fixed image loading of the tutorial - fixes #17863
commit 9ea63ce9 (2024-12-15): BIM: Fixed image loading of the tutorial - fixes #17863
closed 2024-11-18 by yorikvanhavre