FreeCAD/FreeCAD #17865: Solid Cone is open to 360°

juanp1291, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part workbench when created a CONE and you make zoom in a sewing line,  you can measure a little gap (0.0007 mm). (Picture 1 and 2)  
This issue disturb the thickness work of superior face.   (Picture 3)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears.backup1731266494.2616622 1.2.0 (Disabled)
  * fasteners.backup1731266491.981111 0.5.21 (Disabled)
  * sheetmetal.backup1731266497.4283006 0.4.14 (Disabled)
  * A2plus.backup1731266484.1749918 0.4.64a (Disabled)
  * A2plus 0.4.68
  * fasteners 0.5.30
  * freecad.gears 1.3.0
  * Help 1.0.3
  * sheetmetal 0.5.5
```

### Subproject(s) affected?

Part

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
Please check in RC4 or latest weekly

juanp1291, 2024-11-14:
In version: 1.1.0dev.39176 (Git) Conda AppImage. The cone is Ok. Thanks.

closed 2024-11-14 by luzpaz