FreeCAD/FreeCAD #17869: UI: Draft Toggle Construction Mode button size

semhustej, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Draft Toggle Construction Mode button has different height compared to other buttons on the toolbar when using Light or Dark theme. In some positions in the FreeCAD window, the button changes height so that it is aligned with the buttons next to it.

See screenshot:
[image]

------------------
This is my opinion, but the contrast between the tool turned on and off is very small and is not easy to spot at a fast glance. Coloring the button such as Draft Snaps are would be nice in my opinion.

On:
[image]

Off:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-15:
This is caused by theming. Apparently the Light and Dark themes were never checked with the Draft WB active. With the Classic theme the Tray looks OK.

semhustej, 2024-11-15:
> This is caused by theming. Apparently the Light and Dark themes were never checked with the Draft WB active. With the Classic theme the Tray looks OK.
> 
> https://wiki.freecad.org/Draft_Tray

Unfortunately since RC3 there seems to be no Classic theme offered, so Light and Dark are the default ones with this "bug".

Roy-043, 2024-11-15:
The Classic theme can still be selected in the preferences.

Related forum topic:
https://forum.freecad.org/viewtopic.php?t=92104

semhustej, 2024-11-15:
> The Classic theme can still be selected in the preferences.
> 
> Related forum topic: https://forum.freecad.org/viewtopic.php?t=92104

Thanks! I did not know about this.

closed 2024-11-16 by Roy-043

pburger24, 2024-11-18:
@Roy-043 with your commit that you made, The Toggle Construction Mode button is now smaller when using Light/Dark theme and also It it harder to see by looking at the button whether or not it is enabled. Why did you make it smaller?

Roy-043, 2024-11-19:
The button was always intended to be square. The size of the button should match the height of the buttons in the tray (only works for the light and dark theme). With the light theme the depressed state of the button is clear enough I think. For me, the dark theme does not have enough contrast to properly see the difference between the two states of this button (and also of the snap buttons). If other users experience the same then that is something that should be addressed in that theme.

@obelisk79 FYI.

[image]

semhustej, 2024-11-19:
I agree with @Roy-043 

1. In classic mode the button is smaller than the neighboring buttons, so it does not take that much space = better solution. There is no need to have the button wide in my opinion.
2. I also have sometimes problems in seeing enabled buttons in the dark mode due to low contrast. It is not critical though, maybe just getting accustomed to it (up to now I was Classic mode user).
3. From @Roy-043 's screenshot the pressed button in the light theme seems to be more clear than before this commit.
4. Before Roy's commit, there was a problem of the button border staying violet when switched, so user had no overview at all if the tool is on or off, before he/she clicked in the 3D view.

obelisk79, 2024-11-19:
Thanks for the tag @Roy-043.

A couple of comments/questions.

1. Is the toggle construction icon supposed to look like a masonry trawl? I never understood this symbol, I am likely not the only one.
2. The height of these buttons is non-responsive to preferences.
3. Are these hard-coded to a 16px size? They appear proportionate with the rest of the toolbars when I set the icon size to 16px.

I would recommend not hard-coding the icon size to anything which will reside in the toolbars regardless of the rationale. While efficient use of space is important, I believe that visual consistency on a toolbar is more important. I would ask that you at least reconsider this design decision.

Roy-043, 2024-11-20:
@obelisk79 My main reason for tagging you was to bring the contrast issue to your attention. As explained, with the Dark theme for me it is almost impossible to distinguish the 2 states (depressed or not) of this button, and also of the snap buttons. @pburger24 may be affected by this as well.

> 1. Is the toggle construction icon supposed to look like a masonry trawl? I never understood this symbol, I am likely not the only one.

Trowels are used in building construction. For me the reference is clear, but I work in that field.

> 2\. The height of these buttons is non-responsive to preferences.

We are creating QPushButtons and (after my PR) using the height of those to resize the square QToolButton. There is no hard-coded height for QPushButtons in the Draft WB AFAICT.

> would recommend not hard-coding the icon size to anything which will reside in the toolbars regardless of the rationale.

We are not hard-coding the icon size. If the icon in the square button is not readable, then other QPushButtons that include icon would also be unreadable.

PR #17886 (2024-11-16): Draft: fix tray construction mode button for Light and Dark themes
Fixes #17869.
commit 1c19a7e8 (2024-11-16): Draft: fix tray construction mode button for Light and Dark themes (#17886)

Fixes #17869.
commit a5123144 (2024-11-22): Draft: fix tray construction mode button for Light and Dark themes (#17886)

Fixes #17869.