FreeCAD/FreeCAD #17871: Creating a new CAM Job silently fails

stefanboca, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a CAM Job fails and exits after pressing "Ok" on the "Create Job" popup window, with the following error:

```
Job.ERROR: 'OpPrototype' object has no attribute 'setPropertyStatus'
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Main/Gui/Job.py", line 1683, in Create
    obj.ViewObject.Proxy.editObject(obj.Stock)
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Main/Gui/Job.py", line 256, in editObject
    return self.openTaskPanel("Stock")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Main/Gui/Job.py", line 236, in openTaskPanel
    self.taskPanel = TaskPanel(self.vobj, self.deleteObjectsOnReject())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Main/Gui/Job.py", line 754, in __init__
    self.setupOps = PathSetupSheetGui.OpsDefaultEditor(self.obj.SetupSheet, self.form)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in __init__
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in <listcomp>
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 140, in __init__
    self.prototype = op.prototype(name)
                     ^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Base/SetupSheet.py", line 415, in prototype
    self.factory("OpPrototype.%s" % name, ptt)
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Op/Helix.py", line 304, in Create
    obj.Proxy = ObjectHelix(obj, name, parentJob)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Op/Base.py", line 328, in __init__
    self.initOperation(obj)
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Op/CircularHoleBase.py", line 82, in initOperation
    self.initCircularHoleOperation(obj)
  File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Op/Helix.py", line 144, in initCircularHoleOperation
    obj.setPropertyStatus("Direction", ["ReadOnly", "Output"])
    ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'OpPrototype' object has no attribute 'setPropertyStatus'
```

There is no Job Edit task in the task panel after this.

I initially ran into this on 1.0.0RC4. I also tested RC3 and RC2. RC3 had this issue, while RC2 did not.

### Full version info

```shell
OS: Fedora Linux 41 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0RC3.39100 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc3)
Hash: 3c1047103cc8960a92a90a03869a9370b80e6776
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leehambley, 2024-11-14:
I have an extremely similar issue in RC3 and 4. This might be a slight duplicate, but I notice two issues in CAM workbench with similar results to what was originally posted by @stefanboca . My Platform is MACOS arm64 build py311.

When clicking to toggle visibility on "Stock" within "Job" in the Model sidebar

```
pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/IconViewProvider.py", line 86, in setEdit
    self._onEditCallback(True)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/IconViewProvider.py", line 82, in _onEditCallback
    callback(self.obj, self.vobj, edit)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 69, in _OpenCloseResourceEditor
    job.ViewObject.Proxy.editObject(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 256, in editObject
    return self.openTaskPanel("Stock")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 236, in openTaskPanel
    self.taskPanel = TaskPanel(self.vobj, self.deleteObjectsOnReject())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 754, in __init__
    self.setupOps = PathSetupSheetGui.OpsDefaultEditor(self.obj.SetupSheet, self.form)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in __init__
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in <listcomp>
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 140, in __init__
    self.prototype = op.prototype(name)
                     ^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod<class 'AttributeError'>: 'OpPrototype' object has no attribute 'setPropertyStatus'
  ```
  
  And, when clicking "New Job" to add a 2nd side machining operation, a similar but distinct stacktrace:
  
  ```
  23:01:05  Job.ERROR: 'OpPrototype' object has no attribute 'setPropertyStatus'
23:01:05  Traceback (most recent call last):
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 1683, in Create
    obj.ViewObject.Proxy.editObject(obj.Stock)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 256, in editObject
    return self.openTaskPanel("Stock")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 236, in openTaskPanel
    self.taskPanel = TaskPanel(self.vobj, self.deleteObjectsOnReject())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Job.py", line 754, in __init__
    self.setupOps = PathSetupSheetGui.OpsDefaultEditor(self.obj.SetupSheet, self.form)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in __init__
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 221, in <listcomp>
    [OpTaskPanel(self.obj, name, op) for name, op in PathSetupSheet._RegisteredOps.items()],
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/Gui/SetupSheet.py", line 140, in __init__
    self.prototype = op.prototype(name)
                     ^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Base/SetupSheet.py", line 415, in prototype
    self.factory("OpPrototype.%s" % name, ptt)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/Helix.py", line 304, in Create
    obj.Proxy = ObjectHelix(obj, name, parentJob)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/Base.py", line 328, in __init__
    self.initOperation(obj)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/CircularHoleBase.py", line 82, in initOperation
    self.initCircularHoleOperation(obj)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/Helix.py", line 144, in initCircularHoleOperation
    obj.setPropertyStatus("Direction", ["ReadOnly", "Output"])
    ^^^^^^^^^^^^^^^^^^^^^
23:01:05  AttributeError: 'OpPrototype' object has no attribute 'setPropertyStatus'
23:01:05  
  ``` 
  
  Aside the timestamps, the main diff in the second fragment is:
  
  ```
  CAM/Path/Base/SetupSheet.py", line 415, in prototype
[... cut ...]
    self.initOperation(obj)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/CircularHoleBase.py", line 82, in initOperation
    self.initCircularHoleOperation(obj)
23:01:05    File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Op/Helix.py", line 144, in initCircularHoleOperation
    obj.setPropertyStatus("Direction", ["ReadOnly", "Output"])
    ^^^^^^^^^^^^^^^^^^^^^
23:01:05  AttributeError: 'OpPrototype' object has no attribute 'setPropertyStatus'
  ```

maxwxyz, 2024-11-15:
Could you provide an example file? 
@sliptonic @jbaehr FYI

Syres916, 2024-11-15:
Also a report on the forum https://forum.freecad.org/viewtopic.php?p=791987#p791987

jeffgrain, 2024-11-15:
[example.zip](https://github.com/user-attachments/files/17776496/example.zip)
here is an example file.. just load and try to " create job " in the cam workbench

maxwxyz, 2024-11-15:
@sliptonic would you consider this a CAM blocker?

jbaehr, 2024-11-15:
>   File "/tmp/.mount_FreeCAeaElBd/usr/Mod/CAM/Path/Op/Helix.py", line 144, in initCircularHoleOperation
    obj.setPropertyStatus("Direction", ["ReadOnly", "Output"])
    ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'OpPrototype' object has no attribute 'setPropertyStatus'

This line is indeed what I introduced with #17655 (commit 1027926be95f9d6c13e37dbd5abac96ff62e5ece), I'll take a look.

jbaehr, 2024-11-15:
It looks like we're having two *nearly* identical copies of the class `OpPrototype` ([here](https://github.com/FreeCAD/FreeCAD/blob/ed40de3e2390f11b3ce3f5951fc083f954d7d088/src/Mod/CAM/Path/Base/Property.py#L165) and [there](https://github.com/FreeCAD/FreeCAD/blob/ed40de3e2390f11b3ce3f5951fc083f954d7d088/src/Mod/CAM/Path/Base/SetupSheetOpPrototype.py#L179)). They seem to be some stand-in for where usually a document object is passed. They mock a subset of its interface, so a quick fix would be to just add a `setPropertyStatus` method there, too (just like the existing `setEditorMode`).

I'd like to understand more what's going on here, though, and add proper tests to prevent issues like this in the future.

jbaehr, 2024-11-15:
I think I got it. I could reproduce the issue interactively via GUI and was able to create a unit test demonstrating it as well. A proposed fix is here: #17894

dekraftsoftware, 2024-11-16:
Could get a job created with FreeCAD_weekly-builds-39215-conda-Linux-x86_64-py311.AppImage while i had same errors as others on th FreeCAD_1.0.0RC4 build

RodrigoTMorais, 2024-11-17:
I have the same inssue on freecad RC04

stefanboca, 2024-11-19:
Looks like this was fixed, so I'll close the issue.

closed 2024-11-19 by stefanboca