FreeCAD/FreeCAD #17873: Varset PropertyInteger entry limite to 2 digits

macdroid53, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In build 39176.

Create a new Varset PropertyInteger the enter field is restricted to 2 digits.
If the Fx is selected more digits are allowed but closing the Fx popup results in '99' being entered.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenDark/Qt default
Installed mods: 
  * freecad.gears.backup1728167078.8916416 1.3.0 (Disabled)
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * Defeaturing 1.2.2
  * Help 1.0.3
  * kicadStepUpMod 11.4.1
  * Curves 0.6.51
  * DynamicData 2.69.0
  * freecad.gears 1.3.0
  * fasteners 0.5.29
  * sheetmetal 0.5.3
  * freecad.gears.backup1730806425.7568283 1.3.0 (Disabled)
  * CurvedShapes 1.0.12
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-15:
@pieterhijma FYI

kaiwas, 2024-11-15:
this title is more successful than mine)

https://github.com/FreeCAD/FreeCAD/issues/16811

maxwxyz, 2024-11-15:
From #16811 

https://github.com/user-attachments/assets/1c20d562-1834-43c1-8bf5-bac38a222acd

PR #17993 (2024-12-02): Gui: Fix VarSet dialog 2 digit problem
Fixes #17873.

It has been tested on all property types.
closed 2024-12-02 by chennes