FreeCAD/FreeCAD #17875: FCBRepAlgoAPI_Cut Breaks TechDraw Section

WandererFan, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDraw Section view fails to produce a result when using FCBRepAlgoAPI_Cut.  Reverting to BRepAlgoAPI_Cut produces a valid result.

[image]

[image]

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39189 (Git)
Build type: debug
Branch: FCBRepAlgoAPI_Cut
Hash: ee1af2748aa4b97015f2bd7ced41c61a6a0bda3d
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

TechDraw

### Anything else?

[newcontainer_test_CutIssue.FCStd.not.zip](https://github.com/user-attachments/files/17759958/newcontainer_test_CutIssue.FCStd.not.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-11-15:
Things go wrong in DrawViewSection::makeSectionCut around line 513.

```
    for (; expl.More(); expl.Next()) {
        const TopoDS_Solid& s = TopoDS::Solid(expl.Current());
        FCBRepAlgoAPI_Cut mkCut(s, m_cuttingTool);
        // BRepAlgoAPI_Cut mkCut(s, m_cuttingTool);
        if (!mkCut.IsDone()) {
            Base::Console().Warning("DVS: Section cut has failed in %s\n", getNameInDocument());
            continue;
        }
        builder.Add(cutPieces, mkCut.Shape());
    }
```

I haven't dug into FCBRepAlgoAPI_Cut, but it appears that at some point it may not return correctly(??).

WandererFan, 2024-11-15:
ping @AIRCAP

maxwxyz, 2024-11-15:
@WandererFan is the problematic algorithm the default and should this be a blocker?

AIRCAP, 2024-11-15:
looking into it

AIRCAP, 2024-11-15:
trying to debug this, a bit tricky because its an asynchronous process, and the exception is only thrown when the notification comes that the cut is complete. 

I think what happens is that somewhere in FCBrepAlgoAPI_Cut - an exception is thrown because one of the source shapes is not valid, however the subthread in makeSectionCut does never catch exceptions and just gets ended -- DVSCutPieces1.brep is never written because of that.

the old BrepAlgoAPI does not throw an excception if the shapes its supposed to cut are invalid - FCBrepAlgoAPI does --- normally this doesn't matter much because the operation failed anyway, and the exception is eventually caught and displayed 

but due to the asynchronous nature of makeSectionCut - this exception isn't caught where it should be and causes the whole thread to fail, returning no output shape at all (instead of writing a warning Section cut has failed in ...) and still adding the remaining pieces together.

AIRCAP, 2024-11-15:
the uncaught exeption in question is

```gdb
#0  0x00007ffff13a1662 in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fffccbd3dbd in Standard_ConstructionError::Throw() const (this=0x7ffefc48ead0) at /usr/include/opencascade/Standard_ConstructionError.hxx:35
#2  0x00007fffc7c0859f in Standard_ConstructionError::Raise(char const*) (theMessage=0x7fffc7e03b70 "Base shape is not valid for boolean operation")
    at /usr/include/opencascade/Standard_ConstructionError.hxx:35
#3  0x00007fffc7d5e623 in FCBRepAlgoAPI_BooleanOperation::FCBRepAlgoAPI_BooleanOperation(TopoDS_Shape const&, TopoDS_Shape const&, BOPAlgo_Operation)
    (this=0x7fff96791790, theS1=..., theS2=..., theOperation=BOPAlgo_CUT) at ~/src/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:50
#4  0x00007fffc7d5edb1 in FCBRepAlgoAPI_Cut::FCBRepAlgoAPI_Cut(TopoDS_Shape const&, TopoDS_Shape const&) (this=0x7fff96791790, S1=..., S2=...)
    at ~/src/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_Cut.cpp:39
#5  0x00007fffb6dc722f in TechDraw::DrawViewSection::makeSectionCut(TopoDS_Shape const&) (this=0x55555a32b110, baseShape=...) at ~/src/FreeCAD/src/Mod/TechDraw/App/DrawViewSection.cpp:513
#6  0x00007fffb6dc6c57 in TechDraw::DrawViewSection::<lambda()>::operator()(void) const (__closure=0x55555cc9e030) at ~/src/FreeCAD/src/Mod/TechDraw/App/DrawViewSection.cpp:475
#7  0x00007fffb6dcd2e8 in QtConcurrent::StoredFunctorCall0<void, TechDraw::DrawViewSection::sectionExec(TopoDS_Shape&)::<lambda()> >::runFunctor(void) (this=0x55555cc9e010)
    at /usr/include/x86_64-linux-gnu/qt5/QtConcurrent/qtconcurrentstoredfunctioncall.h:70
#8  0x00007fffb6d3ce7f in QtConcurrent::RunFunctionTask<void>::run() (this=0x55555cc9e010) at /usr/include/x86_64-linux-gnu/qt5/QtConcurrent/qtconcurrentrunbase.h:136
#9  0x00007ffff159af82 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff15979d2 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff061c609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007ffff10ba353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

AIRCAP, 2024-11-15:
```C++
if (!BRepCheck_Analyzer(theS1).IsValid()){
        Standard_ConstructionError::Raise("Base shape is not valid for boolean operation");
    }

```

AIRCAP, 2024-11-15:
and here is the culprit:
[image]

AIRCAP, 2024-11-15:
@maxwxyz , @WandererFan  I wouldn't consider this a blocker, since the error is only triggered when sectioning invalid geometry. valid parts should not trigger that. nevertheless, I suggest backporting the fix to 1.0

AIRCAP, 2024-11-15:
I should add - with BRepAlgoAPI, which doesn't do the extra check, OCCT tries to cut the part and in this case would succeed instead of throwing the exception. This however is not guaranteed, doing booleans on invalid parts has undefined outcome - it can turn out fine - or it can segfault OCCT.  I originally didn't have this exception check in the wrapper, it was added as a precaution suggested in https://github.com/FreeCAD/FreeCAD/pull/17119#issuecomment-2405949712

and supported by https://dev.opencascade.org/doc/overview/html/specification__boolean_operations.html#specification__boolean_10_1_1

> **Common requirements**
> 
> Each argument should be valid (in terms of BRepCheck_Analyzer), or conversely, if the argument is considered as non-valid (in terms of BRepCheck_Analyzer), it cannot be used as an argument of the algorithm.
>

AIRCAP, 2024-11-15:
What I just noticed, in 

https://github.com/FreeCAD/FreeCAD/blob/ee1af2748aa4b97015f2bd7ced41c61a6a0bda3d/src/Mod/TechDraw/App/DrawViewSection.cpp#L511-L519

the sub-shapes in the compound are cut individually - this will no longer be necessary if #17530 is ever merged, as the wrapper would automatically do that. That being said, compound traversal should be recursive - as each subshape in a compound could in turn also be a compound, which might fail the cut if it has overlapping shapes.

I'm making a note in #17530 w.r.t. techdraw.

WandererFan, 2024-11-15:
The source shape is from A2+ and we have seen invalid shapes from A2+ before.

WandererFan, 2025-02-20:
This also occurs with TechDraw Detail.  https://forum.freecad.org/viewtopic.php?t=94943

WandererFan, 2025-08-29:
Still relevant, but may become irrelevant shortly.

hyarion, 2025-09-01:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/23102