FreeCAD/FreeCAD #17880: Non-Latin file name is shown as Gibberish

vovodroid, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Follow up to  **STEP import fails #16886** .

After fix in RC4 file is open, but it's name is not displayed correctly:

[Ножки.zip](https://github.com/user-attachments/files/17173891/default.zip)

[image]

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Israel (en_IL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * CurvedShapes 1.0.12
  * MeshRemodel 1.10.32
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdieterm, 2024-11-15:
Have you tried the import setting as described in
https://forum.freecad.org/viewtopic.php?p=791115#p791115
?

vovodroid, 2024-11-15:
Yes, setting UTF-8 helps, thanks! Should it be default choice?

luzpaz, 2024-11-15:
@vovodroid Please open a forum.freecad.org discussion to discuss making this a default feature. Thanks

closed 2024-11-15 by luzpaz