FreeCAD/FreeCAD #17883: No effect on changing Edit / Preferences -- BaseApp/Preferences/Dialog/DontUseNativeDialog

Ventusfahrer, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ubuntu 24.10 / Gnome Desktop:

Regardless the value of **BaseApp/Preferences/Dialog/DontUseNativeDialog**  true/false opening a file will always show the standard Qt-FileDialog and not the current one from the operating system.

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.29

Starting FreeCAD in SafeMode does not change anything.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
@kadet1090 could you take a look at this?

hkienle, 2025-03-10:
For me, it works as expected. (The new settings are picked up immediately, no need to restart FreeCAD.)

OS: KDE **Flatpak** runtime (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/172.811
Installed mods: 
  * Gridfinity 0.9.7
  * STEMFIE 0.3.1
  * freecad.gears 1.3.0
  * Curves 0.6.58

[image]

Ventusfahrer, 2025-03-11:
> For me, it works as expected. (The new settings are picked up immediately, no need to restart FreeCAD.)
> 
> (Edit: Based on past experience, I would not be surprised if it is (still) broken with AppImage, but works with Flatpak.)
> 
> OS: KDE **Flatpak** runtime (GNOME/gnome/wayland) Architecture: x86_64 Version: 1.1.0dev.40267 (Git) Flatpak Build type: Release Branch: (HEAD detached at [04d2bcb](https://github.com/FreeCAD/FreeCAD/commit/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd)) Hash: [04d2bcb](https://github.com/FreeCAD/FreeCAD/commit/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd) Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1 Locale: English/United States (en_US) Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/ Logical/physical DPI: 96/172.811 Installed mods:
> 
> * Gridfinity 0.9.7
> * STEMFIE 0.3.1
> * freecad.gears 1.3.0
> * Curves 0.6.58
> 
> [image]

I may have expressed myself in a misleading way:

I did not mean that changing the setting of the parameter is not stored in the FreeCAD settings but changes have no effect on the FileOpenDialog.

The Qt-Implementation of the FileDialog on Ubuntu looks like this:

[image]

The Operating System FileDialog (called Native Dialog) looks like this:

[image]

The difference on Ubuntu Gnome is, that you have the possibilty to use a Favorit list which makes navigation much more efficient.

The difference is in calling the FileDialog in PyQt6:

```
       file_dialog = QFileDialog(self)
       selected_file = file_dialog.getOpenFileName(self, options=QFileDialog.Option.DontUseNativeDialog)
```

does show the Qt-specific one,

```
       selected_file = file_dialog.getOpenFileName(self)
```

does show the nativ one.

In the App-Image Development version of FreeCAD I see no difference in the FileOpen Dialog regardless the settings of above parameter. It always shows the Qt-specific FileDialog.

hkienle, 2025-03-11:
> new settings are picked up immediately, no need to restart FreeCAD

I am afraid I did not community this cleary: I meant to say, if you change DontUseNativeDialog from true to false (or vice versa) and then FileOpenDialog you immediately see that the the change. (This is perhaps obvious, but I did expect that I would have to do a restart of FreeCAD to take effect.)

closed 2025-09-13 by github-actions[bot]