FreeCAD/FreeCAD #17885: RC4 Appimage can't find fusermount

pyro9, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Ubuntu 18.04.6 LTS

When running FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage it reports: 
fuse: failed to exec fusermount: Permission denied

Worked around with ln -s /bin/fusermount /usr/bin/fusermount

RC2 works and previous works without the workaround.

### Full version info

```shell
after workaround:
OS: Ubuntu 18.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * ThreadProfile 1.90.0
  * Curves 0.6.51
  * 3D_Printing_Tools
  * lattice2 1.0.0
  * Silk 1.0.0
  * Assembly4 0.50.15
  * A2plus 0.4.68
  * CfdOF 1.21.6
  * fcgear.backup1677967119.4973533 1.0.0 (Disabled)
  * fcgear 1.0.0
  * sheetmetal 0.2.61
  * Render 2023.8.18
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-15:
@adrianinsaval maybe you know what's going on here

adrianinsaval, 2024-11-15:
appimage bug, I'm hoping they will merge the fix soon or I'll compile teh tooling with the fix before final release: https://github.com/AppImage/type2-runtime/pull/37