FreeCAD/FreeCAD #17890: PartDesign: Pad causes Geom2dConvert Concatenation Error for a complex body

YohDeadfall, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For a part made from a curve based solid I cannot apply the Pad operation while FreeCAD correctly renders it. It shows `Geom2dConvert Concatenation Error` message in the log. That happens only when the upper face of the pad lies on the same plane as another face produced by a Pocket operation.

It can be easily reproduced by changing the `Length` property of `Pad001` to `4.5`. Or that pad can be removed and another one created from `Sketch007` using the same length.

[The `repro.FCStd.zip` file](https://github.com/user-attachments/files/17780451/repro.FCStd.zip) is a normal FreeCAD file, but with the `ZIP` extension add to prevent GitHub from complaining about TGZ. So, just rename it to ``repro.FCStd`. Sorry for inconvenience.

[image]

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.3.12
  * A2plus 0.4.60n
  * Curves 0.6.48
```

### Subproject(s) affected?

PartDesign

### Anything else?

The base body is the same as in #17682, so the root cause might be the same.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
@YohDeadfall can you please try with RC4 or latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@AIRCAP FYI

AIRCAP, 2024-11-17:
That part is heavily broken. When you do body generation with complex curve generated shapes, you should do geometry checks at every step. Your part already has continuity warnings at the base feature and already an invalid edge at the first pad "Pad" .. from there things go downhill. You then do a fillet - which already causes a whole page full of invalid geometry errors.

I cannot reproduce the actual reported issue with latest main btw...

maxwxyz, 2024-11-17:
I'm closing this. @YohDeadfall please comment if you can reproduce the issue on latest RC or weekly with a clean file

closed 2024-11-17 by maxwxyz

YohDeadfall, 2024-11-17:
Thanks, I will give it a try later today or tomorrow.

As it was mentioned in the linked issue, the first warning happens with the first fillet. Why does it happen and how to avoid such issues?

YohDeadfall, 2024-11-18:
@tomate44, I followed the advice by @AIRCAP and now I see that the geometry check for the `Rotation_Sweep` generates errors `Boolean operation: GeomAbs_C0` for two edges and a face. To see that `Run boolean operator checks` needs to be turned on. Does it mean that it goes to the Curves repo, or is it me doing something wrong?

@maxwxyz, yeah, I still can repro it with the latest weekly.

maxwxyz, 2024-11-18:
We need to be sure that the same issue is also existing in an error-free model

YohDeadfall, 2024-11-18:
Sorry, I'm not following. What do you mean?

tomate44, 2024-11-20:
I don't see any bug involved in CurvesWB tools that are used here.
The 2 JoinCurves that are used to build the RotationSweep are C0 continuous.
To force higher continuity, user can activate Approximation in JoinCurves, with MaxDegree = 3, and adjust Nb Samples at a reasonable value (~10 seems good enough)
Next problem is the Ruled Surface used to close the RotationSweep on the XZ plane.
RuledSurface generates a BSplineSurface, not a real Part.Plane.
So the Refine op. of the first Pad will fail and the following fillet fails to generate a valid result around the thin picky areas of this BSplineSurface.

YohDeadfall, 2024-11-21:
Does it mean that I'm doing something unsupported and another way should be fine do design it?