FreeCAD/FreeCAD #17891: QPixmap::setMask() mask size differs from pixmap size

JohnOCFII, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the current weekly, I'm noticing this item shows up repeatedly in the Report View when I launch FreeCAD to the Startup Page:

`QPixmap::setMask() mask size differs from pixmap size`

I'm not seeing any impact in my designs -- but wanted to note this.

[image]

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-11-16:
Also happens in RC4.  Launching Preferences seems to cause the issue.  

```
OS: macOS 15.1
Architecture: arm64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```

[image]

maxwxyz, 2024-11-16:
@furgo16 @cjmayo could this be related to the latest PRs? https://github.com/FreeCAD/FreeCAD/pull/15601 https://github.com/FreeCAD/FreeCAD/pull/17151 ?

furgo16, 2024-11-16:
My PR touched bitmap sides, but on the document tree, which is not shown on the Start page. Unfortunately, I'm not familiar enough with Qt to tell where the size check in the warning comes from.

maxwxyz, 2024-11-16:
@kadet1090 maybe?

xtemp09, 2024-11-16:
No such messages.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39220 (Git)
Build type: Debug
Branch: main
Hash: 38bf51c76809fe7eb77a5b4d49d7b572471d4fb7
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

JohnOCFII, 2024-11-16:
This was on a 2021 MacBook Pro 16" with effective screen resolution of: 1728x1117.

JohnOCFII, 2024-12-25:
More info.  I recently deleted my `user.cfg` and `system.cfg` for a "clean start" and noticed these messages only started when I installed and switched to the OpenThemes. Maybe @obelisk79 will have an idea.

```
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39741 (Git) Conda
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.34
  * Curves 0.6.52
```

[image]

JohnOCFII, 2025-08-29:
I still see this regularly.

luzpaz, 2025-08-29:
ping @obelisk79