FreeCAD/FreeCAD #17892: Linux AppImage broken since somewehere beween RC3 on some maschines

caco3, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have 2 Ubuntu machines where I run freeCAD.
On one maschine, all AppImages start correctly. But on the other machine, the starts fails since RC3:
```bash
_> ./FreeCAD_1.0.0RC3-conda-Linux-x86_64-py311.AppImage 
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/gruinelli/appImages/FreeCAD_1.0.0RC3-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/gruinelli/appImages/FreeCAD_1.0.0RC3-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory
```
RC2 still is ok:
```bash
_> ./FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

User path has changed to /home/gruinelli/.local/share/FreeCAD/. Please move user modules and macros
The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
```

I copied the bad images to the other machine and they run fine there, so the files are ok.

Both machines are x86_64 and have the latest Ubuntu 24.04 updates:
```bash
> lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:        24.04
Codename:       noble
```

### Full version info

```shell
Versions see above.
The `--safe-mode` does not provide more information.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2024-11-15:
The RC3 and RC4 images use the `zstd` compression, which is the default for the `appimagetool`.  It is somewhat surprising to see that Ubuntu 24.04 does not support this very popular compression method.

caco3, 2024-11-15:
Thanks for the hint.
I investigated further and found this issue: https://github.com/TheAssassin/AppImageLauncher/issues/602#issuecomment-2460297857.
Indeed the "bad" machine had the "appimagelauncher" installed. After I removed it, also RC3 and 4 work.

caco3, 2024-11-15:
And it also already got reported in https://github.com/FreeCAD/FreeCAD-Bundle/issues/331

closed 2024-11-15 by caco3

oursland, 2024-11-15:
Thanks for following-up. AppImageLauncher has been a problem for us as it is no longer keeping up with AppImage specs.

Was AppImageLauncher installed by default?

caco3, 2024-11-15:
No, I think I installed it because I would like to have a GUI to start and update the images.