FreeCAD/FreeCAD #17895: Assembly WB Moving an item by one particular end freezes FreeCAD

philjynx, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have an assembly with two sliding items. Grabbing either end and moving it has the expected outcome.
The other one (which is essentially very similar) can be moved by grabbing one end but attempting to do so from the other end breaks FreeCAD.

This applies to 1RC2, 1RC4 and WB 39176.
You can see that here.
https://youtu.be/a5IBxRpmTN8

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear
  * FreeCAD-themes 2024.7.24
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.4.1
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-11-16:
@philjynx In my opinion, you should upload the project here to test it on other computers and debug it.

maxwxyz, 2024-11-16:
Please share the example file and steps how to reproduce. You can also upload videos directly to the comment.

philjynx, 2024-11-16:
@maxwxyz I just tried to drop the video here
[image]

philjynx, 2024-11-16:
@maxwxyz The model I posted was relatively complicated (by my standards) using a varset to control all its components. I've created a far simpler version with no varset, fixed values for all constraints and no external files. 

**This one also behaves erratically**, sometimes grabbing a part and trying to move it causes it to 'explode' other times it simply doesn't move.

Oddly, when I first created it, I could grab any part of the moveable section and move it. After a save and a restart it has become flakey.

Two of the sketches have an oddity in their properties, as per this screen shot. 
[image]
[Simplified.zip](https://github.com/user-attachments/files/17784955/Simplified.zip)

philjynx, 2024-11-16:
I have removed the YouTube video as I have no interest in promoting a negative attitude to FreeCAD.

huguesdpdn-aerospace, 2024-11-17:
# Test
I tested. I have 0 knowledge about the Assembly workbench but here is the test output under Linux with the latest build:
```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

# Video
[The video in high definition on my Google Drive](https://drive.google.com/file/d/1U2jwIW8E9TSeNmz0Qlkb-yp6qwkp_rKd/view?usp=sharing)

# Analysis
From what I can see, only the center one and one side of the cross rails works, not the other side (right side working not left -  bottom side working but not top)
Even weirder, if I click on the left block, the horizontal "rail" is switching/flipping (see video) on the other side, not for the vertical bar.

# Output
No crash
No error message
No error in the Python console
No freeze

# Question
@philjynx : Did I test it wrong? Should I test in another way to get the freeze ?

philjynx, 2024-11-18:
@huguesdpdn-aerospace 
My knowledge if assembly WB is zero too, this thread is the result of my first attempt at using assembly WB.

I don't see how you could test it wrongly, the whole point is that the 'joints' _should_ allow only movement along the outside rails. 

Alas, there us currently no documentation that clearly explains how to use the various types of joint.

I suspect that the creators of FreeCAD may be very experienced in using other, commercial, CAD software.

I am not!

Well done getting an appimage to run, I've tried to no avail on my Linux Mint machine....

MiqCG, 2024-12-03:
Hi,
Definetly not an expert here but I have been testing the assembly WB for months. I have tested your Simplified file in 1.0.0 and 1.1.0 revision 39416 with no problems (crash, errors or freezing). What am not able to do is drag the central component in the X axis; I can only do it dragging by the bar. Maybe we can consider this an issue.
I attach a video if I am missing something. 

On the other hand, FYI, the assembly can work with two less joints (16 in total), and two fixed joints instead of two slider joints. I attach the modified file.

[Simplified_modified.zip](https://github.com/user-attachments/files/17994801/Simplified_modified.zip)

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

https://github.com/user-attachments/assets/278f1001-d3c9-493e-a1a6-fba461987073

philjynx, 2024-12-04:
@MiqCG Thanks for having a look at it. 
There is one Part that I've noticed in the original file (and therefore your modified version) which should not be there it is 'ThroughBlock002'.

Removing (deleting) it does not alter the behaviour of the model.

I just downloaded my own file and tested it, it did not behave badly.
Mysterious.

MiqCG, 2024-12-04:
This type of mysteries are annoying.
Are you using a different revision than RC4 (39104)? The dragging could have been solved in the meanwhile for other reasons.
Have you tested also the original file (not the simplified version that you attached) that gave you problems?

closed 2025-07-18 by maxwxyz