FreeCAD/FreeCAD #17899: Part Design Helix has an issue when a Taper Angle is specified

gryphonB-2, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to construct a tapered thread, I encountered this problem, which I hope is demonstrated in the accompanying file.
In sum: 
It is possible to construct a tapered thread, ie a sketch profile lofted through a tapered helix. But it is not then possible to add any further geometry. 
Neither is it possible to create some geometry & then add a tapered thread to it. 
My solution was to use a Part Helix and loft the profile through that, which works just fine. 
This appears not related to my use of the DynamicData WB or anything else I can see. 
While it's an issue from version 1 RC1, I do not see it reported elsewhere, so include it now. 

**Note: The attached FC file has not been zipped, just renamed. To open, just rename, removing the .zip extension**
[Silicon Nozzles2.FCStd.zip](https://github.com/user-attachments/files/17783686/Silicon.Nozzles2.FCStd.zip)

[image]
[image]
[image]
 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.5.3
  * Curves 0.6.51
  * DynamicData 2.69.0
  * slic3r-tools
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
@gryphonB-2 is it still an issue in RC4? @FlachyJoe FYI

FlachyJoe, 2024-11-16:
@gryphonB-2 the given file doesn't contain the bug demo. Please provide a minimal failing file.

gryphonB-2, 2024-11-23:
Hi luzpaz, thanks for the response. As stated earlier, the uploaded .zip file is in fact a FCStd file, with the extension renamed. I have deleted the example I created for this, so I'd ask you check yourself with that file if issue persists in RC4.
Cheers. Appreciate your work.

tarman3, 2024-11-23:
Dont understand the issue.
Can you attach file with issue and write instruction (step by step)?

gryphonB-2, 2024-11-29:
[PartDesign-HelixTaperBug2.FCStd.zip](https://github.com/user-attachments/files/17952947/PartDesign-HelixTaperBug2.FCStd.zip)

Prelim:
The included files, I hope, demonstrate the bug. 
As we all know, the FCStd file is just a zip file with another extension. The demonstration file IS A FCStd FILE WITH A '.zip' EXTENSION ADDED. RENAME THE FILE, REMOVING THE '.zip' EXTENSION. THE FILE WILL OPEN JUST FINE.

To recreate:
1. Profile on XZ plane, on X axis, sketch of a triangle tilted 5 deg. 
		Helix is created in +ve Z direction.
[image]

2. Positive Cone Angle specified. 
		Helix does not change, apart from the cone angle causing radius to grow
[image]

gryphonB-2, 2024-11-29:
[PartDesign-HelixTaperBug2.FCStd.zip](https://github.com/user-attachments/files/17953028/PartDesign-HelixTaperBug2.FCStd.zip)

Prelim:
The included files, I hope, demonstrate the bug. 
AS we all know, the FCStd file is just a zip file with another extension. The demonstration file IS A FCStd FILE WITH A '.zip' EXTENSION ADDED. RENAME THE FILE, REMOVING THE '.zip' EXTENSION. THE FILE WILL OPEN JUST FINE.
The file was created in the latest version: FreeCad 1.0

Caveat: Weirdly, even though the included file shows the bug, I have a copy which does not: the tapered Helix creates correctly in PartDesign & fuses with the shell without the thread disappearing. But since I've spent days trying to model the part & hours making the files to demonstrate the bug, I'm jacked with the whole thing. If this doesn't explain it then nothing will & we should all consider new careers.
1. Profile on XZ plane, on X axis, sketch of a triangle tilted 5 deg. 
		Helix is created in +ve Z direction.
[image]

2. Positive Cone Angle specified. 
		Helix does not change, apart from the cone angle causing radius to grow

[image]

3. Negative Cone Angle Specified
		Helix is now created in the negative Z direction
		Radius also grows. 
		Technically, the angle is negative, but is not desired or expected.
[image]

4. Helix Reversed is specified
		Result is same as if a positive angle is specified, not reversed
[image]

5. Helix with sketch to be revolved as shell for the thread
		The thread is clearly visible.
[image]

6. Shell Sketch revolved about Z axis.
		Thread clearly disappears. 
[image]

Hope that clears up any confusion. 
		
Now, lets do the same thing, in Part:
1. Recreate geometry, because any reuse will literally drive you insane. 
		Same thread profile, same helix.
[image]

2. Sweep ThreadProfile along Helix
		Helix creates as expected, tapering upwards, and not flipping directions as in PartDesign
		Sweep completes quickly. 
[image]

3. Sketch for surrounding shell
[image]

4. Revolution & Fuse
		Sketch003 is revolved about Z axis, intersecting with Sweep
		The two objects are Fused. 
		Operation completes quickly & accurately
[image]

I sincerely hope this clears up any confusion you may have experienced in my previous bug report. I could of course spend another two hours recreating this bug for you, although Im sure that anyone with even minimal understanding of FreeCad could do this for themselves in a few minutes. In other words, if you cant understand this, then you never will. There is literally nothing more I could do or say to demonstrate this. Nothing. 

And just to repeat: as we all know, the FCStd file format is simply a zip file. Hence there is no reason to compress it, except that GitHub wont accept it. Therefore, the included file IS a FCSTD file. JUST REMOVE THE '.zip' EXTENSION, RETAINING THE  'FCStd'. The file will open just fine.

tarman3, 2024-11-29:
Diplicate
https://github.com/FreeCAD/FreeCAD/issues/18108

maxwxyz, 2024-11-29:
Closing as duplicate then

closed 2024-11-29 by maxwxyz