FreeCAD/FreeCAD #17900: Invalid container exception on Undo/redo leads to crash

furgo16, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Uncompress the [test file](https://github.com/user-attachments/files/17268821/TestAssembly.zip) and open it with FreeCAD
2. Select the `Assembly002` object on the document tree and delete it with the <kbd>Del</kbd> key
3. Undo the previous action (<kbd>Ctrl</kbd>+<kbd>Z</kbd> or click the `Std_Undo` command on the toolbar)
4. Redo the previously undone action (<kbd>Ctrl</kbd>+<kbd>Y</kbd> or click the `Std_Redo` command on the toolbar)
- **Expected**: the redo command works and the `Assembly002` object is removed (we're back at step 2 being done)
- **Actual**: the redo command does not work, the `Assembly002` object is not removed, and the user is presented with this error:

```
<Exception> invalid container
<App> Transactions.cpp(186): Exception on redo 'Delete':invalid container
```

Invariably, after this error, FreeCAD crashes at some point. Not immediately, but after doing a few operations.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39216 (Git) Snap 1193
Build type: Release
Branch: main
Hash: ed40de3e2390f11b3ce3f5951fc083f954d7d088
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
@PaddleStroke could you take a look at this if you'd consider this a blocker?

luzpaz, 2024-11-16:
@maxwxyz imho any undo/redo bugs are considered high priority since they really can screw up a users productivity. Do you agree ?

maxwxyz, 2024-11-16:
@luzpaz yes

MiqCG, 2024-11-17:
I can confirm the steps of furgo16, Redo doesn't work, either the assembly container is active or not. On the other hand, I didn't get the crashing (I just moved the parts around)

https://github.com/user-attachments/assets/c963aaf9-b94b-45ce-aa7b-a8d8488a9877

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

huguesdpdn-aerospace, 2024-11-18:
# Test

I did a test ... and yes, I got the Exception AND a crash (SIGSEGV Segmentation fault).

## CTRL+Y do nothing: **CONFIRMED**

## Manual Python command: `Gui.runCommand('Std_Redo',0)` : **CONFIRMED**
```
<Exception> invalid container
<App> Transactions.cpp(186): Exception on redo 'Delete':invalid container
```

## Crash: Segfault:  **CONFIRMED**
We do not get the segfault "after doing a few operations", but only when I close FreeCAD. If I do not press the `DEL` key on the Assembly object, FreeCAD exit properly.
```
FreeCAD 1.1.0, Libs: 1.1.0devR39215 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

<Exception> invalid container
<App> Transactions.cpp(186): Exception on redo 'Delete':invalid container
<Exception> invalid container
<App> Transactions.cpp(186): Exception on redo 'Delete':invalid container
Program received signal SIGSEGV, Segmentation fault.
/tmp/.mount_FreeCAIdDgHj/AppRun : line 31 : 173459 Erreur de segmentation  (core dumped) "${MAIN}" "$@"
```
(yes, I tested twice the Assembly delete so you have 2 times the exception, I was expecting FreeCAD to crash when doing 
"a few operations".
After SEGFAULT, FreeCAD returned code **139** (`echo $?`)

# Video

https://github.com/user-attachments/assets/b5a3a87d-be84-44ef-8ce7-dfcbe6fbe5bc

# FreeCAD version:
```
OS: Ubuntu 24.04.1 LTS (LXQt/Lubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39215 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b3a2268289482c65c18bbb4cc073b33020e71ce2
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

PaddleStroke, 2024-11-18:
Doing the procedure : "Delete + Undo + Redo" has different results in different cases : 

- Case 1 : App::Part or Assembly::Object with NO joints lots of errors, but it works and does not crash : 

> 10:08:17  Origin "Unnamed#Origin" doesn't contain feature with role "XY_Plane"
> 10:08:17  Can't find Origin for "Unnamed#Assembly"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "X_Axis"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "X_Axis"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "X_Axis"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "X_Axis"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XZ_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XY_Plane"
> 10:08:18  Origin "Unnamed#Origin" doesn't contain feature with role "XY_Plane"
> 10:08:18  No view provider linked to the Origin
> 10:08:18  No view provider linked to the Origin

Case 2 : AssemblyObject with 1 joint : 

> 10:11:00  Origin "Unnamed#Origin" doesn't contain feature with role "XY_Plane"
> 10:11:00  Can't find Origin for "Unnamed#Assembly"
> 10:11:02  <Exception> invalid container
> 10:11:02  <App> Transactions.cpp(187): Exception on redo 'Delete':invalid container

Then crash.

So I'm not sure what is going on here, but it seems delete/undo/redo has a bug outside of assembly, and that jointObjects are crashing due to this.

PaddleStroke, 2024-11-18:
After further test I can confirm that this is not an Assembly bug. It's a core bug.
Here is a minimal file reproducing the bug :
[delete-undo-redo.zip](https://github.com/user-attachments/files/17797809/delete-undo-redo.zip)

[image]

PaddleStroke, 2024-11-18:
Actually the Part is not even necessary. A standalone link can reproduce this.
[image]

PaddleStroke, 2024-11-18:
@wwmayer this is a core bug/crash concerning undo/redo if that interest you.

huguesdpdn-aerospace, 2024-11-18:
Linked to #17913 too - Same crash, same exception just different crash stacktrace

PR #17954 (2024-11-25): Core: Do not detach object from document before the call of breakDepe…
…ndency

This fixes #17900
commit eb3ce44b (2024-11-25): Core: Do not detach object from document before the call of breakDependency

This fixes #17900
commit a15134d7 (2024-11-25): Core: Do not detach object from document before the call of breakDependency

This fixes #17900
closed 2024-11-25 by yorikvanhavre