FreeCAD/FreeCAD #17901: Sketcher:Dimension display does not display  single digit

rjwilson01, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When displaying dimensions
15m is displayed as 15.0 
anything from 1.0 mto 9.999 m is displayed as mm
No change is observed when changing number of decimals or NumberFormat in edit preferences
Lacking better control over the automatic change in units the change in display from m to mm should occur at 0.9999 mm not at 9.9999 mm

### Full version info

```shell
[code]
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
Please test on latest 1.0 RC4 version or weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

rjwilson01, 2024-11-16:
Just to be sure that I replicated this on the weekly build correctly - 

- I downloaded FreeCAD_weekly-builds-39176-conda-Windows-x86_64-py311.7z  - 
- Unzipped it to a folder and moved the folder to C:\Program Files\FreeCAD 0.21\FreeCAD_weekly-builds-39176-conda-Windows-x86_64-py311  to get it into the secure area 
- I then ran  C:\Program Files\FreeCAD 0.21\FreeCAD_weekly-builds-39176-conda-Windows-x86_64-py311\bin\FreeCad.exe - 
- Loaded the old file - it asked if it should recalculate - I said yes
- Still the same - I deleted and readded a constaint as well (it is the 1,000.0mm one not lined up but next to the 4,000mm and 5,000mm in the png)
- [image]

I ask as I did not uninstall the old one  - so if there is some path problem you should probably tell me how to woraround it

Help About 
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5

Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

maxwxyz, 2024-11-16:
That's ok, you can just extract it in your Downloads folder and run the FreeCAD.exe inside the bin folder. No need to uninstall anything. To remove just delete the folder again. Thanks for testing.
What units are you using (preferences)? Is it m or mm? and how many decimal places is your default?

rjwilson01, 2024-11-16:
[image]
MKS - meters (m) 
I tried a few decimal places 0 to 2 - but 2 is how I have it set  
I also tried a each of the Number format settings with no effect on existing contraints - I did not delete and readd them to tes this though

Roy-043, 2024-11-16:
This has nothing to do with Sketcher, this is just how the MKS unit system behaves. The fact that 'm' units are not displayed depends on a Sketcher preference however.

See:
"Hide base length units for supported unit systems"
https://wiki.freecad.org/Sketcher_Preferences#Display

The issue title seems wrong to me.

> Lacking better control over the automatic change in units the change in display from m to mm should occur at 0.9999 mm not at 9.9999 mm

This also seems incorrect.

maxwxyz, 2024-11-16:
So is this intended behavior or what is the bug then?

Roy-043, 2024-11-16:
That the MKS unit system shows any value below 10m in mm is awkward IMO. But I have to assume that this is intended. I personally prefer a fixed length unit: FEM (mm) or Meter decimal.

Both MKS and the Standard unit system have this issue. I would suggest closing this and opening a new issue instead.

rjwilson01, 2024-11-17:
I see this issue is now closed - I have no idea of what to do which would be useful - you've closed an issue with information saying a new issue should be raised  - but on what should the new issue be raised The SI MKS system description ? and am I meant to copy what I found ?

This does seem related to https://github.com/FreeCAD/FreeCAD/issues/5906 In that the behaveour in 5906 is explained by what I found but I an see no way to contribute any further 

> This has nothing to do with Sketcher, this is just how the MKS unit system behaves. 
Are you saying its a MKS system as in code in the FreeCAD app , or are you claiming the MKS units specification says you should not display single initeger digits?
I checked https://physics.nist.gov/cuu/Units/checklist.html it seems silent on the topic

You also state "Standard" unit  has the same issue do you mean

> The fact that 'm' units are not displayed depends on a Sketcher preference however.
When is m units not displayed? - everywhere in my system the m is displayed  Your comment is unrelared to the behaviour I have reported
> 
> See: "Hide base length units for supported unit systems" https://wiki.freecad.org/Sketcher_Preferences#Display
> 
> The issue title seems wrong to me.
Wrong is unhelpful what should it be?
> 
> > Lacking better control over the automatic change in units the change in display from m to mm should occur at 0.9999 mm not at 9.9999 mm
> 
> This also seems incorrect.
Incorrect is not useful - what should it be?

Roy-043, 2024-11-18:
The title of this issue is confusing. It suggests that single digit dimensions do not show a numerical value at all.

The confusion about  "Hide base length units for supported unit systems" was triggered by: 

> 15m is displayed as 15.0

Apparently you meant "15 m is displayed as 15.0 m"

> display from m to mm should occur at 0.9999 mm not at 9.9999 mm

The mentioned values should be  0.9999 m and 9.9999 m (in m not in mm).

As you yourself mention this issue is closely related to #5906. I'll as a comment there.