FreeCAD/FreeCAD #17903: Assembly: turning subassembly to rigid changes its orientation

furgo16, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Uncompress the [test file](https://github.com/user-attachments/files/17783800/test-exploded-view-sub-new_rigid.zip) and open it with FreeCAD
2. Double click on the `Assembly` object on the document tree to activate the assembly
3. Click on the Insert component command (`Assembly_InsertLink`) on the toolbar
4. On the Tasks panel dialog, Ensure `Rigid sub-assemblies` is unchecked
5. On the Tasks panel dialog click on the `Assembly002` object to insert it
6. **Notice**: the inserted object does not show a transform gizmo to move it around, as it generally happens with other inserted components
7. On the Tasks panel dialog, press `Ok`
8. Press `Ctrl+R` to recompute
9. **Notice**: the following warnings appears on the Report view
   ```test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
      test_exploded_view_sub_new_rigid#Fixed004 is unconnected to a grounded part so it is ignored.
      1.327e-06 <App> Document.cpp(2903): test_exploded_view_sub_new_rigid#Fixed004 still touched after recompute
   ``` 
10. On the document tree, right click on the `Assembly002` object
11. On the context menu shown, choose the `Turn rigid` command at the top
- **Expected**: the sub-assembly is turned rigid without further modifications
- **Actual**: the sub-assembly is rotated 90° along the X axis

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39216 (Git) Snap 1193
Build type: Release
Branch: main
Hash: ed40de3e2390f11b3ce3f5951fc083f954d7d088
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Anything else?

This is a test file originally created as a test case for another bug. I do not discard that there might have been errors in creating it that might lead to the issue. But it being a small model, I hope they are easy to spot if they are indeed there.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-16:
Cannot confirm, it is not changing orientation on my side.
I still get the same warnings about the fixed constraint.
[image]

@PaddleStroke FYI

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

MiqCG, 2024-11-17:
I can confirm a change in orientation when turning non-rigid to rigid, but in my case is a flip (180º). After the flip, the dragging behaves weirdly.

https://github.com/user-attachments/assets/a441f629-ae99-48f6-8169-ca74e47c5550

Note: the "control-z" at the end of the video were attempts of crashing, because FreeCAD crashed with a control-z in a previous test.

MiqCG, 2024-11-17:
PS: if it helps, I attach the files. They are just a random group of joined components for testing purposes only. Open Assembly.FCStd file.

[Subassemblies.zip](https://github.com/user-attachments/files/17789121/Subassemblies.zip)

VincidaB, 2024-12-03:
Investigating this (using @MiqCG 's file), with me the behavior is different if you first `solve Assembly` and then turn the sub assembly rigid (but the sub assembly joint get's flipped and it gets selected for some reason)

With the original file, `objLinkMap` is empty and looks to be the source of the link between the upper assembly and the sub assembly.

https://github.com/FreeCAD/FreeCAD/blob/77ba6eed4f1b007b9b0d0883e7432866e5fee381/src/Mod/Assembly/App/AssemblyLink.cpp#L108-L120

Should this map be empty or should is it cleared at some point by error ?

furgo16, 2025-01-17:
I've added some additional information with a screenshot and screencast on the original description. I tested this issue again on the latest daily and it's still reproducible. What I noticed is that,

1. when I add the (sub-)assembly to the main assembly, it loses its grounded joint. This is probably because there can only be a grounded joint per assembly, and the main assembly already has one.
2. The components of the inserted (sub-)assembly at this point are all over the place, probably because of the missing grounded joint and the fact that the main assembly is not yet connected to it => NOK
3. If I then connect the main assembly to it with a fixed joint, the (sub-)assembly appears displaced => NOK

PaddleStroke, 2025-07-21:
> Notice: the base Assembly001 object has a grounded and a fixed joint. When inserting it as Assembly002, the linked object is missing the grounded joint.

This is intended. You don't want the sub-assembly to be grounded or you could not put it where you want.