FreeCAD/FreeCAD #17904: Can't activate an external Body contained in a Part in a App::Link until it is expanded manually.

PaddleStroke, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This may sound obscure but this problem is blocking an assembly feature I'm currently making.

Steps to reproduce : 
- Create new document 'Part.fcstd' and save it. (doc A)
- Add a App::Part and inside it a Body.
- Create 2nd new document and save it (doc B)
- In doc B create a link.
- set the link linked object to the App::Part in doc A
- Now try to activate body in doc A 
`Gui.ActiveDocument.ActiveView.setActiveObject('pdbody', App.getDocument('Part').getObject('Body'))`
It doesn't work!

- Now expand the Link, which shows body.
- Collapse link again.
- Try to activate again : It works.

In video :

https://github.com/user-attachments/assets/0530b1c0-4230-4ebf-8acd-4ccbeb732af6

@wwmayer any idea why this is happening?

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39188 (Git)
Build type: Release
Branch: asm_new_part
Hash: 4bc8cd60e9a9c24ed007f0d3d341ca70498c798a
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.60k
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * fasteners 0.5.0
  * FreeCAD - Shortcut.lnk
  * freecad.gears 1.2.0
  * FreeCAD_FastenersWB-master 0.4.56
  * kicadStepUpMod 10.23.0
  * Ondsel 2024.9.13.02
  * OpenDark 2023.12.17
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-11-16:
For now I have made a work-around function to expand the link before trying to activate the body 
```
    def expandLinkManually(self, link):
        mw = Gui.getMainWindow()
        trees = mw.findChildren(QtGui.QTreeWidget)

        Gui.Selection.addSelection(link)
        for tree in trees:            
            items = tree.selectedItems()
            for item in items:
                tree.expandItem(item)
```

wwmayer, 2024-12-01:
> any idea why this is happening?

I guess that at this point the tree item of the body link is not created yet

FEA-eng, 2025-10-26:
@PaddleStroke Can you confirm if it's still relevant? I can't reproduce the issue. Is that because you already implemented that workaround?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-11-03:
I assume that it's still relevant, but maybe @PaddleStroke could reconfirm.

PaddleStroke, 2026-11-03:
still happening in main. Fixed in astocad [6364fb4](https://github.com/AstoCAD/FreeCAD/commit/6364fb4beae8e47b81c3cb2510d9798bd1e3ce77) (feel free to PR/port any commit from AstoCAD whenever you want)