FreeCAD/FreeCAD #17905: Windows7: 1.0RC-4 fails to start.

CorvusCorax, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This was reported on Discord:

https://discord.com/channels/870877411049357352/870877411623985154/1307274496075370527

Victor_UK
 — 
Today at 10:22 AM

> Just downloaded RC4 as there is  now a windows installer.
> Installs OK on my Win7 machine...
> But fails to start due to a missing api-ms-win-core-path-|1-1-0.dll ☹️
> [The windows .7z weekly builds also give the same error.  Hopefully this will be fixed for full release] 

### Full version info

```shell
1.0 RC4 on Windows 7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-16:
The FreeCAD packagers are no longer supporting Windows 7 hence it doesn't work, the nearest you can get is to check https://github.com/kisolre/FreeCAD/releases/tag/weekly_Win7 and see when the next build is produced but this is being done on a voluntary basis so no complaints if they stop being updated. The proper way forward is to create a dual boot with Linux Mint 22 IMHO which is the path I took a few years ago.

closed 2024-11-16 by maxwxyz

VictorLDunn, 2024-11-16:
I understand that older versions of Windows (and other OS's) become unsupported over time, but, there is currently no indication on the FreeCAD download page, or the "Help us squashing last-minute bugs" and "Download and try a 1.0 Release Candidate build!" link to RC4, that Windows 7 is **not** supported.